openwrt/packages #1957: mjpg_streamer: "Trace/breakpoint trap" crash on ramips with trunk

roger-, 2015-11-12:
Been testing trunk with a WT3020 (ramips mt7620) and I noticed that mjpg_streamer seems to be broken.

Launching it with ` mjpg_streamer -i "input_uvc.so -d /dev/video0"` causes a crash:

> Trace/breakpoint trap 

I haven't had time to investigate, but perhaps it's related to the recent musl change?

thess, 2015-11-14:
I'll see what I can do in the next couple of days or so. Is this problem only on the ramips platform. I'll have to build an image later today for the one ramips (rt305x) device I have. 

I regularly use mjpg_streamer on an armel (ixp4xx nslu2) platform and have noticed no problems using the trunk & 3.18 kernel, etc.

roger-, 2015-11-14:
Thanks Ted.

Unfortunately I only have access to raimps devices right now, but I'll have to see if I can replicate it on something else. Gonna see if I can get strace running for now.

thess, 2015-11-14:
Build from trunk - works OK for me on a rt305x platform (3.18 kernel is the default). mjpg_streamer does segfault while shutting down but I don't know how old that problem is. My guess is your problem is specific to the mt76xx platform?

roger-, 2015-11-15:
strace log:

'''
ioctl(4, _IOC(_IOC_READ|_IOC_WRITE, 0x56, 0x1a, 0x50), {index=0, name="Camera 1", type=V4L2_INPUT_TYPE_CAMERA}) = 0
ioctl(4, VIDIOC_G_FMT, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, fmt.pix={width=640, height=480, pixelformat=v4l2_fourcc('M', 'J', 'P', 'G'), field=V4L2_FIELD_NONE, bytesperline=0, sizeimage=614400, colorspace=V4L2_COLORSPACE_SRGB}}) = 0
ioctl(4, VIDIOC_ENUM_FMT, {index=0, type=V4L2_BUF_TYPE_VIDEO_CAPTURE, flags=0, description="YUV 4:2:2 (YUYV)", pixelformat=v4l2_fourcc('Y', 'U', 'Y', 'V')}) = 0
--- SIGTRAP {si_signo=SIGTRAP, si_code=SI_USER, si_errno=ENETUNREACH, si_pid=0, si_uid=0} ---
+++ killed by SIGTRAP +++
Trace/breakpoint trap
```

Looks like someone else was having the same issue here: http://pastebin.com/FzcT5cPf

Not sure if this is a related issue: https://github.com/openwrt/packages/issues/1692

thess, 2015-11-16:
Perhaps more info is to be gained by building mjpg_streamer with -DDEBUG to get its own trace ouput. You may get a better idea of where in the code you are dying.  SIGTRAP is not normal unless there exists poorly generated code, messed up stack or other general corruption of an image.

You might want to try building gdbserver into your system and use `scripts/remote-gdb` to see if you can catch the trap. To fully debug, follow the outline in: http://wiki.openwrt.org/doc/devel/gdb

roger-, 2015-11-16:
I'll have a go at `-DDEBUG` to see what's going on before pulling out gdb, thanks.

@jow -- any other ideas?

thess, 2015-11-18:
Found it! Patch coming shortly in PR.

It is the memcpy on line 133 in v4l2uvc.c - should be `sizeof(struct v4l2_fmtdesc)`

roger-, 2015-11-18:
Nice work Ted! How'd you figure that out?

thess, 2015-11-19:
Fresh config/build and got something to fail. A few extra DBG trace statements and narrowed it down to the memcpy. Finally noticed that the target was larger than the source struct and the memcpy was trashing the data previously calloc'd (and referencing beyond the source struct size).

thess, 2015-11-20:
Closed by #1987

closed 2015-11-20 by thess

roger-, 2015-11-22:
Seems to work now  Ted, thanks!