openwrt/packages #1960: xl2tpd port change bug

psyborg55, 2015-11-15:
when trying to change port of my L2TP connection the server side of the xl2tpd is listening on new port, but the port change has no effect on client side resulting in client trying to connect on default 1701 port.

yousong, 2015-11-16:
How did you do the port change?  The code indicates that `port` option in `default` section is not for this.  Looks like it's a per LAC setting.  Have you tried specifying OpenWrt server option with port?  e.g. `option server example.com:1702`

psyborg55, 2015-11-16:
changed the port in xl2tpd.conf on a server (VPS ubuntu 12.04) and the server is started on a new port. OpenWrt is acting as a client and the same xl2tpd.conf port change had no effect regardless of the port option being global. since i'm using luci to configure connections i'm not sure how to specify this option server variable to affect only the port but not host being used. this is pure L2TP test scenario without IPSEC.

yousong, 2015-11-16:
I see.  The validation in LuCI requires that fields to be of type hostname and it will reject values like `example.com:1702`.  Most likely you need to ssh or telnet into the box and manually edit the `server` option.

the `port = nnn` in `[global]` section of xl2tpd.conf is intended as a server (lns) option only.  It's not about which port to connect when acting as a lac.

jow-, 2015-11-16:
maybe it makes sense to abstract that difference away though

yousong, 2015-11-16:
Hmm, sorry, the above method won't work.  As the protocol script currently also assumes that server is a hostname and refuse to setup the interface if resolveip fails...

I will cook up a patch for this

yousong, 2015-11-16:
Can you give the following patch a try.  Test it by manually editting `/lib/netifd/proto/l2tp.sh` should also work

```
diff --git a/net/xl2tpd/files/l2tp.sh b/net/xl2tpd/files/l2tp.sh
index 8a931ad..c0a227f 100644
--- a/net/xl2tpd/files/l2tp.sh
+++ b/net/xl2tpd/files/l2tp.sh
@@ -25,14 +25,15 @@ proto_l2tp_init_config() {
 proto_l2tp_setup() {
 	local interface="$1"
 	local optfile="/tmp/l2tp/options.${interface}"
+	local host
 
 	local ip serv_addr server
-	json_get_var server server && {
-		for ip in $(resolveip -t 5 "$server"); do
-			( proto_add_host_dependency "$interface" "$ip" )
-			serv_addr=1
-		done
-	}
+	json_get_var server server
+	host="${server%:*}"
+	for ip in $(resolveip -t 5 "$host"); do
+		( proto_add_host_dependency "$interface" "$ip" )
+		serv_addr=1
+	done
 	[ -n "$serv_addr" ] || {
 		echo "Could not resolve server address" >&2
 		sleep 5
```

psyborg55, 2015-11-16:
this works! thanks a lot for making this patch. i've already looked through the same l2tp.sh file myself but had no idea how to add something like this.

now the only thing that prevents setting the desired port from luci is field validation which doesn't allow anything except hostname or ip address. is it possible to make luci accept port number for example 0.0.0.0:11000 or even add another field for port?

yousong, 2015-11-17:
Glad that it works.  I will submit a revised patch with another option `port` and close this issue now.

@jow- I'd like to keep xl2tpd as it is since changing it may introduce unexpected effects for current users.

closed 2015-11-17 by yousong

yousong, 2015-12-02:
Hello, @psyborg55 

Just to let you know that now specifying port in `server` option is possible and supported by LuCI.  Big thanks go to @jow-

psyborg55, 2015-12-03:
ok, thank you both for the assistance. i'm currently using l2tp on different port and no problems so far. btw is there a changeset listed on dev board regarding this?