openwrt/packages #1983: libgpg-error: compile fails with gcc-5.2.0

joaochainho, 2015-11-18:
Hi,
Since trunk r47495, libgpg-error fails to compile with gcc-5.2.0.
Tested after 'make dirclean'.
```
mips-openwrt-linux-musl-gcc -E -I/home/OpenWrt/trunk/staging_dir/target-mips_24kc_musl-1.1.11/usr/include -I/home/OpenWrt/trunk/staging_dir/target-mips_24kc_musl-1.1.11/include -I/home/OpenWrt/trunk/staging_dir/toolchain-mips_24kc_gcc-5.2.0_musl-1.1.11/usr/include -I/home/OpenWrt/trunk/staging_dir/toolchain-mips_24kc_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/OpenWrt/trunk/staging_dir/toolchain-mips_24kc_gcc-5.2.0_musl-1.1.11/include   _mkerrcodes.h | grep GPG_ERR_ | \
               gawk -f ./mkerrcodes.awk >mkerrcodes.h
rm _mkerrcodes.h
cc -I. -I. -o mkerrcodes ./mkerrcodes.c
In file included from ./mkerrcodes.c:26:0:
./

   { , "GPG_ERR_E2BIG" },
     ^
./mkerrcodes.h:10:5: error: expected expression before ',' token
   { , "GPG_ERR_EACCES" },
     ^
./mkerrcodes.h:11:5: error: expected expression before ',' token
   { , "GPG_ERR_EADDRINUSE" },
     ^
./mkerrcodes.h:12:5: error: expected expression before ',' token
   { , "GPG_ERR_EADDRNOTAVAIL" },
     ^
./mkerrcodes.h:13:5: error: expected expression before ',' token
   { , "GPG_ERR_EADV" },
     ^
./mkerrcodes.h:14:5: error: expected expression before ',' token
   { , "GPG_ERR_EAFNOSUPPORT" },
     ^
./mkerrcodes.h:15:5: error: expected expression before ',' token
   { , "GPG_ERR_EAGAIN" },
     ^
./mkerrcodes.h:16:5: error: expected expression before ',' token
   { , "GPG_ERR_EALREADY" },
     ^
./mkerrcodes.h:17:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADE" },
     ^
./mkerrcodes.h:18:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADF" },
     ^
./mkerrcodes.h:19:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADFD" },
     ^
./mkerrcodes.h:20:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADMSG" },
     ^
./mkerrcodes.h:21:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADR" },
     ^
./mkerrcodes.h:22:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADRQC" },
     ^
./mkerrcodes.h:23:5: error: expected expression before ',' token
   { , "GPG_ERR_EBADSLT" },
     ^
./mkerrcodes.h:24:5: error: expected expression before ',' token
   { , "GPG_ERR_EBFONT" },
     ^
./mkerrcodes.h:25:5: error: expected expression before ',' token
   { , "GPG_ERR_EBUSY" },
     ^
./mkerrcodes.h:26:5: error: expected expression before ',' token
   { , "GPG_ERR_ECANCELED" },
     ^
./mkerrcodes.h:27:5: error: expected expression before ',' token
   { , "GPG_ERR_ECHILD" },
     ^
./mkerrcodes.h:28:5: error: expected expression before ',' token
   { , "GPG_ERR_ECHRNG" },
     ^
[... 332 lines cut ...]
make[5]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/po'
Making all in lang
make[5]: Entering directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/lang'
Making all in cl
make[6]: Entering directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/lang/cl'
echo '@errnos@' | cat ../../src/err-codes.h.in - ../../src/errnos.in \
	| gawk -f ./mkerrcodes.awk >gpg-error-codes.lisp
make[6]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/lang/cl'
make[6]: Entering directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/lang'
make[6]: Nothing to be done for `all-am'.
make[6]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/lang'
make[5]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/lang'
make[5]: Entering directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12'
make[5]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12'
make[2]: *** [/home/OpenWrt/trunk/build_dir/target-mips_24kc_musl-1.1.11/libgpg-error-1.12/.built] Error 2
make[2]: Leaving directory `/home/OpenWrt/trunk/feeds/packages/libs/libgpg-error'
make[1]: *** [package/feeds/packages/libgpg-error/compile] Error 2
make[1]: Leaving directory `/home/OpenWrt/trunk'
make: *** [package/libgpg-error/compile] Error 2
```
TIA

hauke, 2015-11-18:
This is fixed in #1990 I am just waiting for the maintainer to approve.

hauke, 2015-11-20:
fixed in  #1990

closed 2015-11-20 by hauke

joaochainho, 2015-11-20:
Thanks! It compiles fine now.