openwrt/packages #1993: glib2 not builds because of missing `usr/lib/libffi-3.0.13/include`

g1itch, 2015-11-19:
When I trying to build image from the latest buildroot it fails with the following messages:
```#sh
...
make[3]: Entering directory '/home/system/openwrt/feeds/packages/libs/glib2'
make[3]: Leaving directory '/home/system/openwrt/feeds/packages/libs/glib2'
make[3]: Entering directory '/home/system/openwrt/feeds/packages/libs/libffi'
rm -rf /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libffi
mkdir -p /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/stamp /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libffi
install -d -m0755 /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libffi/usr/lib
cp -fpR /home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/libffi-3.0.13/ipkg-install/usr/lib/libffi.so.* /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libffi/usr/lib/
SHELL= /home/system/openwrt/staging_dir/host/bin/flock /home/system/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libffi/. /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/'
rm -rf /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libffi
touch /home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/stamp/.libffi_installed
make[3]: Leaving directory '/home/system/openwrt/feeds/packages/libs/libffi'
make[3]: Entering directory '/home/system/openwrt/feeds/packages/libs/glib2'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/include  -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/include -I/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/include -I/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/include  -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/include -I/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/include -I/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/include " LDFLAGS="-L/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib -L/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/lib -L/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/lib -L/home/system/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/lib -znow -zrelro -L/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/lib -L/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/lib " make   -C /home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-stub/include -I/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-stub/include" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1'
make  all-recursive
make[5]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1'
Making all in .
make[6]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1'
make[6]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1'
Making all in m4macros
make[6]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/m4macros'
make  all-am
make[7]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/m4macros'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/m4macros'
make[6]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/m4macros'
Making all in glib
make[6]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib'
make  all-recursive
make[7]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib'
Making all in libcharset
make[8]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/libcharset'
make  all-am
make[9]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/libcharset'
make[9]: Nothing to be done for 'all-am'.
make[9]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/libcharset'
make[8]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/libcharset'
Making all in gnulib
make[8]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/gnulib'
make  all-am
make[9]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/gnulib'
make[9]: Nothing to be done for 'all-am'.
make[9]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/gnulib'
make[8]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/gnulib'
Making all in pcre
make[8]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/pcre'
make  all-am
make[9]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/pcre'
make[9]: Nothing to be done for 'all-am'.
make[9]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/pcre'
make[8]: Leaving directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/pcre'
Making all in update-pcre
make[8]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/update-pcre'
make  all-am
make[9]: Entering directory '/home/system/openwrt/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/glib/update-pcre'
make[9]: Nothing to be done for 'all-am'.
[... cut ...]
make[3]: Leaving directory '/home/system/openwrt/feeds/packages/libs/glib2'
package/Makefile:191: recipe for target 'package/feeds/packages/glib2/compile' failed
make[2]: *** [package/feeds/packages/glib2/compile] Error 2
make[2]: Leaving directory '/home/system/openwrt'
package/Makefile:188: recipe for target '/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/home/system/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/system/openwrt'
```

hauke, 2015-11-20:
see this fix: #1989

g1itch, 2015-11-21:
It iseams this patch solves this issue but glib2 still not builds. Now I have the following messages:
```sh
...
Making all in tests
make[5]: Entering directory '/home/sandbox/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/tests'
make  all-recursive
make[6]: Entering directory '/home/sandbox/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/tests'
Making all in gobject
make[7]: Entering directory '/home/sandbox/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/tests/gobject'
make[7]: *** No rule to make target '/home/sandbox/openwrt/trunk/staging_dir/host/bin/glib-genmarshal', needed by 'stamp-testmarshal.h'.  Stop.
make[7]: Leaving directory '/home/sandbox/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/glib-2.44.1/tests/gobject'
Makefile:1645: recipe for target 'all-recursive' failed
```

Do I need to create the separate issue for it?

thess, 2016-08-07:
Stale issue - re-submit if still a problem.

closed 2016-08-07 by thess

sunke1988, 2017-02-08:
@g1itch , what is the status of your last issue "No rule to make target '/home/sandbox/openwrt/trunk/staging_dir/host/bin/glib-genmarshal', needed by 'stamp-testmarshal.h'"
I meet the same problem on my ubuntu PC when I build glib-2.38.2.
Can you give some clues ? thanks