openwrt/packages #2005: libffi: patch failed

wongsyrone, 2015-11-22:
Hi, the patch is still broken.

```diff
--- a/libffi.pc.in
+++ b/libffi.pc.in
@@ -2,10 +2,10 @@ prefix=@prefix@
 exec_prefix=@exec_prefix@
 libdir=@libdir@
 toolexeclibdir=@toolexeclibdir@
-includedir=${libdir}/@PACKAGE_NAME@-@PACKAGE_VERSION@/include
+includedir=@includedir@
 
 Name: @PACKAGE_NAME@
 Description: Library supporting Foreign Function Interfaces
 Version: @PACKAGE_VERSION@
-Libs: -L${toolexeclibdir} -lffi
-Cflags: -I${includedir}
+Libs: -lffi
+Cflags:
--- a/include/Makefile.am
+++ b/include/Makefile.am
@@ -5,5 +5,5 @@ AUTOMAKE_OPTIONS=foreign
 DISTCLEANFILES=ffitarget.h
 EXTRA_DIST=ffi.h.in ffi_common.h
 
-includesdir = $(libdir)/@PACKAGE_NAME@-@PACKAGE_VERSION@/include
+includesdir = $(includedir)
 nodist_includes_HEADERS = ffi.h ffitarget.h
--- a/include/Makefile.in
+++ b/include/Makefile.in
@@ -314,7 +314,7 @@ top_srcdir = @top_srcdir@
 AUTOMAKE_OPTIONS = foreign
 DISTCLEANFILES = ffitarget.h
 EXTRA_DIST = ffi.h.in ffi_common.h
-includesdir = $(libdir)/@PACKAGE_NAME@-@PACKAGE_VERSION@/include
+includesdir = $(includedir)
 nodist_includes_HEADERS = ffi.h ffitarget.h
 all: all-am
 
```

hauke, 2015-11-22:
What problem do you have? You problem description "broken" is a little bit short

wongsyrone, 2015-11-22:
@hauke  Sorry. The patch I post above is for my fork (version 3.2). Closing this issue.

closed 2015-11-22 by wongsyrone