openwrt/packages #2020: mwan3-2.0-0: ipv6 dependencies not checked and provide a way to disable ipv6 support

braveheartleo, 2015-11-25:
I noticed that mwan3 was upgraded to `2.0-0`. Unfortunately, this version bump did not check for ipv6 dependencies, thus when I upgraded mwan3 from `1.6-3` in my custom Designated Driver build where ipv6 support is disabled and packages not included, mwan3 failed to start.

The package needs to be modified so that ipv6 dependencies are checked before installation.

I would like to request that mwan3 provide workarounds to turn off ipv6 for non-ipv6 setups.

Adze1502, 2015-12-17:
Thank you for noticing, will get back soon with update

diizzyy, 2017-06-07:
@Adze1502 
Fixed?

feckert, 2017-11-03:
@hnyman i think we could close this issue.
Its very old.

closed 2017-11-03 by hnyman

feckert, 2017-11-03:
@hnyman thanks