openwrt/packages #2024: can't compile btrfs-progs

DocMAX, 2015-11-26:
make[3]: Entering directory '/home/docmax/openwrt/trunk/feeds/packages/utils/btrfs-progs'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/usr/include -I/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/include -I/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/usr/include -I/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/usr/include -I/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/include -I/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/usr/include -I/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/include " LDFLAGS="-L/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/usr/lib -L/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/lib -L/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/usr/lib -L/home/docmax/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.2.0_musl-1.1.11/lib -znow -zrelro " make   -C /home/docmax/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.11/btrfs-progs-v3.19.1/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/home/docmax/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.11/btrfs-progs-v3.19.1'
    [CC]     props.o
props.c:20:24: fatal error: attr/xattr.h: No such file or directory
compilation terminated.
Makefile:149: recipe for target 'props.o' failed
make[4]: *** [props.o] Error 1
make[4]: Leaving directory '/home/docmax/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.11/btrfs-progs-v3.19.1'
Makefile:65: recipe for target '/home/docmax/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.11/btrfs-progs-v3.19.1/.built' failed
make[3]: *** [/home/docmax/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.11/btrfs-progs-v3.19.1/.built] Error 2
make[3]: Leaving directory '/home/docmax/openwrt/trunk/feeds/packages/utils/btrfs-progs'
package/Makefile:191: recipe for target 'package/feeds/packages/btrfs-progs/compile' failed
make[2]: *** [package/feeds/packages/btrfs-progs/compile] Error 2
make[2]: Leaving directory '/home/docmax/openwrt/trunk'
package/Makefile:188: recipe for target '/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/home/docmax/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/docmax/openwrt/trunk'
/home/docmax/openwrt/trunk/include/toplevel.mk:177: recipe for target 'world' failed
make: *** [world] Error 2

mstorchak, 2015-11-26:
Please vote for #1934

DocMAX, 2015-11-28:
hello?

mstorchak, 2015-11-29:
should be fixed by 021c3e119874ba8b7b88b53f69f6b286cbe524d9, please check

DocMAX, 2015-11-29:
its fixed, thx

closed 2015-11-29 by DocMAX