openwrt/packages #2031: luci-app-statistics: clicking graph doesn't attach host/timespan to URL

mattventura, 2015-11-27:
For example:
I have multiple hosts sending their collectd data to a single OpenWRT box. 
I browse to Statistics > Graphs > Interfaces. 
Select desired host in the dropdown, switch to that host. 
Now, the menu bar displaying the different interfaces changes to those of the new host. These links point to the correct place (e.g. /admin/statistics/graph/interface/tun0/?host=some-other-host&timespan=1day). 
However, clicking the individual graphs does not, as their URL lacks the host argument (e.g. /admin/statistics/graph/interface/tun0). 
If the interface does not exist for the default host, then it will result in the "No page is registered at ..." error page. 
On top of that, it doesn't put the desired timespan in the URL, so that will get reset to the default as well. 

Found on 15.05.

diizzyy, 2017-06-07:
@mattventura 
Is still present in LEDE?

hnyman, 2017-06-07:
I close this as clicking the graphs has been disabled in LuCI (and the issue belongs to LuCI repo in any case).

closed 2017-06-07 by hnyman