openwrt/packages #2043: libshout-full build error (library needed to build oggfwd)

build000, 2015-11-28:
```
(...)
make[3]: Entering directory `/home/openwrt/trunk/feeds/packages/libs/speex'
rm -rf /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libspeex
mkdir -p /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/stamp /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libspeex
install -d -m0755 /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libspeex/usr/lib
cp -fpR /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/speex-1.2rc1/ipkg-install/usr/lib/libspeex.so.* /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libspeex/usr/lib/
SHELL= /home/openwrt/trunk/staging_dir/host/bin/flock /home/openwrt/trunk/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libspeex/. /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/'
rm -rf /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libspeex
touch /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/stamp/.libspeex_installed
make[3]: Leaving directory `/home/openwrt/trunk/feeds/packages/libs/speex'
(...)
make[3]: Entering directory `/home/openwrt/trunk/feeds/packages/libs/libtheora'
rm -rf /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libtheora
mkdir -p /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/stamp /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libtheora
install -d -m0755 /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libtheora/usr/lib/
cp -fpR /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/libtheora-1.1.1/ipkg-install/usr/lib/*.so* /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libtheora/usr/lib/
SHELL= /home/openwrt/trunk/staging_dir/host/bin/flock /home/openwrt/trunk/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libtheora/. /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/'
rm -rf /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/tmp-libtheora
touch /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/root-ar71xx/stamp/.libtheora_installed
make[3]: Leaving directory `/home/openwrt/trunk/feeds/packages/libs/libtheora'
(...)
make[3]: Entering directory `/home/openwrt/trunk/feeds/packages/libs/libshout'
mkdir -p /home/openwrt/trunk/bin/ar71xx/packages /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/libshout-nospeex/libshout-2.3.1/ipkg-ar71xx/libshout/CONTROL /home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/pkginfo
install -d -m0755 /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/libshout-nospeex/libshout-2.3.1/ipkg-ar71xx/libshout/usr/lib
cp -fpR /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/libshout-nospeex/libshout-2.3.1/ipkg-install/usr/lib/libshout.so.* /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/libshout-nospeex/libshout-2.3.1/ipkg-ar71xx/libshout/usr/lib/
find /home/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/libshout-nospeex/libshout-2.3.1/ipkg-ar71xx/libshout -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package libshout is missing dependencies for the following libraries:
libspeex.so.1
libtheora.so.0
make[3]: *** [/home/openwrt/trunk/bin/ar71xx/packages/packages/libshout_2.3.1-2_ar71xx.ipk] Error 1
make[3]: Leaving directory `/home/openwrt/trunk/feeds/packages/libs/libshout'
make[2]: *** [package/feeds/packages/libshout/compile] Error 2
make[2]: Leaving directory `/home/openwrt/trunk'
make[1]: *** [/home/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/trunk'
make: *** [world] Błąd 2
```

thess, 2015-11-28:
Fixed by https://github.com/openwrt/packages/commit/40c040abd4886f7b2013bb6c7cfa5c4af1f1d9f6

build000, 2015-11-28:
confirm - this patch  https://github.com/openwrt/packages/commit/40c040abd4886f7b2013bb6c7cfa5c4af1f1d9f6 fixing problem - packages builds clean and properly.

closed 2015-11-28 by build000