openwrt/packages #2044: perl IO Socket problem

RussellSenior, 2015-11-28:
I am seeing a message in logread:

Bareword "Errno::EINVAL" not allowed while "strict subs" in use at /usr/lib/perl5/5.22/IO/Socket.pm line 129

This seems to be related to gcc-5.2, see:

  http://www.nntp.perl.org/group/perl.perl5.porters/2015/09/msg230796.html

@Naoir

RussellSenior, 2015-11-29:
With
CONFIG_GCC_VERSION="4.8-linaro"
no error.

Naoir, 2015-11-29:
This all boils down to the configuration system not setting 'gccversion' correctly, which Errno uses to adapt to GCC5's behaviour. It'll then silently fail to parse the error code definitions, resulting in Errno::EINVAL(and friends) not being defined as described by you.

I'll push a fix to master shortly.

Thank you a lot!

commit bb065396 (2015-11-29): perl: Set gccversion configuration symbol correctly

This was previously manually set to '4.8.0'. Aside from just being incorrect,
it also breaks Errno's GCC5 detection.

Fixes #2044.
closed 2015-11-29 by Naoir