openwrt/packages #2099: vpnc: mtu is set incorrectly

sudden6, 2015-12-10:
The automatically set mtu of 1500 for the vpn interface is incorrect, as the mtu on the vpn interface must be smaller than the mtu of the physical interface. Most commonly a the mtu of the vpn interface is set 88 smaller, though I don't know where this value comes from.

Additionally it isn't possible to manually set the mtu with
```
option mtu '1412'
```

I solved this by replacing 
```
if [ -n "$INTERNAL_IP4_MTU" ]; then
		MTU=$INTERNAL_IP4_MTU
	fi

	if [ -z "$MTU" ]; then
		MTU=1412
	fi
``` 

in /lib/netifd/vpnc-script with

```
ip link set dev $TUNDEV mtu 1412
```

but I'm not sure if this is the right way.

yousong, 2016-08-13:
ping maintainer of vpnc @danielg4

danielg4, 2016-08-13:
Ping @nmav too, since he is the maintainer of the vpnc-script package…

nmav, 2016-08-25:
The how much smaller depends on the VPN protocol and its overhead (IP+UDP+TLS for example). Yes the 88 is arbitrary and most likely is due to PPP. For a DTLS-based VPN such as openconnect the overhead will be a little more to 113 bytes for IPv4. Should we make it 1360 to cover any possible combinations?

sudden6, 2016-08-25:
I'm not using this setup anymore, but at that time I would have preferred documentation about this overhead and the `option mtu '1412'` to actually change something so I could manually set the optimal value.

nmav, 2017-04-29:
I am not sure we can do anything here. Please open a pull request if you have some concrete proposal.

closed 2017-04-29 by nmav

zolg, 2024-11-11:
Faced the same issue in 2024.
In my opinion the most non-intrusive way to solve it is to create a /etc/openconnect/post-connect.d/set_mtu script with the following contents:
`#!/bin/sh
if [ -n "$INTERNAL_IP4_MTU" ]; then
	ip link set $TUNDEV mtu $INTERNAL_IP4_MTU
fi`
Although this solution does not allow setting an arbitrary mtu in /etc/config/network, but it respects the mtu negotiated with the server and survives system updates.

commit 4034e9ee (2017-06-18): openconnect: new option mtu

According to openconnect --help output:

  -m, --mtu=MTU                   Request MTU from server
      --base-mtu=MTU              Indicate path MTU to/from server

Fixes #2099 by allowing setting tunnel mtu
commit b91c48ba (2017-06-18): openconnect: new option mtu

According to openconnect --help output:

  -m, --mtu=MTU                   Request MTU from server
      --base-mtu=MTU              Indicate path MTU to/from server

Fixes #2099 by allowing setting tunnel mtu