openwrt/packages #2111: libffi: Fails to build

ClaymorePT, 2015-12-11:
The setup is the following:
- GCC 5.2
- binutils 2.25.1
- glibc 2.21

>libtool: compile:  mips-openwrt-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I.. -I. -I../include -Iinclude -I../src -I/home/claymore/Desktop/openwrt/staging_dir/target-mips_34kc_glibc-2.21/usr/include -I/home/claymore/Desktop/openwrt/staging_dir/target-
mips_34kc_glibc>-2.21>/include -I/home/claymore/Desktop/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_glibc-2.21/usr/include -I/home/claymore/Desktop/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_glibc-2.21/include -I. -I../include -Iinclude -I../src -Os -pipe -mno-branch->likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT src/mips/o32.lo -MD -MP -MF src/mips/.deps/o32.>Tpo -c ../src/mips/o32.S  -fPIC -DPIC -o src/mips/.libs/o32.o
>../src/mips/o32.S: Assembler messages:
>../src/mips/o32.S:94: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f12,0*4($sp)'
>../src/mips/o32.S:101: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f12,0*4($sp)'
>../src/mips/o32.S:109: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f12,0*4($sp)'
>../src/mips/o32.S:110: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f14,2*4($sp)'
>../src/mips/o32.S:115: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f12,0*4($sp)'
>../src/mips/o32.S:116: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f14,1*4($sp)'
>../src/mips/o32.S:123: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f12,0*4($sp)'
>../src/mips/o32.S:124: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f14,2*4($sp)'
>../src/mips/o32.S:131: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f12,0*4($sp)'
>../src/mips/o32.S:132: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f14,2*4($sp)'
>../src/mips/o32.S:161: Error: opcode not supported on this processor: mips32r2 (mips32r2) `s.s $f0,0($8)'
>../src/mips/o32.S:168: Error: opcode not supported on this processor: mips32r2 (mips32r2) `s.d $f0,0($8)'
>../src/mips/o32.S:265: Error: opcode not supported on this processor: mips32r2 (mips32r2) `s.d $f12,((14*4)-10*4)($fp)'
>../src/mips/o32.S:266: Error: opcode not supported on this processor: mips32r2 (mips32r2) `s.d $f14,((14*4)-8*4)($fp)'
>../src/mips/o32.S:285: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f0,((14*4)-6*4)($fp)'
>../src/mips/o32.S:289: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f0,((14*4)-6*4)($fp)'
>Makefile:1108: recipe for target 'src/mips/o32.lo' failed
>make[5]: *** [src/mips/o32.lo] Error 1

Here is the [output](https://www.dropbox.com/s/9mqek784zsf40s7/libffi-fails.txt?dl=0)

ClaymorePT, 2015-12-14:
For reference: https://github.com/atgreen/libffi/issues/187

srdjanrosic, 2015-12-27:
risk@...
    $ make package/libffi/compile
     make[1] package/libffi/compile
     make[2] -C package/libs/toolchain compile
     make[2] -C feeds/packages/libs/libffi compile
    risk@...
    $ echo $?
    0

can you still reproduce this?

ClaymorePT, 2015-12-28:
I believe this problem was already solved some time in the last month. I
can no longer reproduce it.
Thank you anyway!

closed 2016-01-10 by thess