openwrt/packages #2117: Vanilla install of LXC + luci-lxc-app not working

tcuthbert, 2015-12-13:
Hi,

I have pulled the luci-lxc-app packages from the opkg repos but unable to get an openwrt container running successfully.

When I attempt to attach to a busybox container I get an error that an error that the init pid is incorrect.

Is there any documentation on how to set LXC up on openwrt?

vancluever, 2016-01-16:
Just thought I'd mention something: There seems to have been no activity on this package in a year, and also the domain referenced here has now been squatted and redirects to what looks like sketchy stuff.

```
feeds/packages/utils/luci-app-lxc/files/lxc.config:6:	option url 'virtualwrt.org/containers/'
```

Is it safe to assume this package is dead?

vladwing, 2017-09-19:
I just tried this package. The virutalwrt.org url is completely dead.

localnet, 2017-11-24:
I did a pull request to luci-app-lxc for it use images.linuxcontainers.org.

If somebody want to test it: https://github.com/openwrt/luci/pull/1422

dibdot, 2018-08-18:
Fixed in 18.06

closed 2018-08-18 by dibdot