openwrt/packages #2131: Package bridge-utils missing in CC 15.05

damian-kaczkowski, 2015-12-15:
For description please read here.
https://dev.openwrt.org/ticket/21387

jow-, 2015-12-15:
Yes it is but I'm not sure if it will be backported. You can use iproute2 instead.

damian-kaczkowski, 2015-12-15:
Could you then please package oldpackages the old way? I mean like the packages were always packaged for 10.03, 12.09, 14.07?

eg: http://git.openwrt.org/?p=14.07/packages.git;a=tree

This way we won't miss packages when they will be moved to github. oldpackages just worked on trunk, and they should just work on 15.05 too. We shouldn't loose them just because moving them from one location to another imo.

diizzyy, 2017-06-02:
@hnyman @thess
Close this, 15.05 is unsupported and frozen.

closed 2017-06-02 by hnyman