openwrt/packages #2141: Python 2.7.11 Build Breakage

psidhu, 2015-12-17:
With the new bump to Python 2.7.11, I'm seeing the following breakage:

```
ImportError: No module named _sysconfigdata_nd
configure: error: python2.7 interpreter not found
make[3]: *** [/usr/src/psidhu/openwrt/openwrt-next/build_dir/target-arm_cortex-a9+neon_musl-1.1.11_eabi/Python-2.7.11/.configured_yy
yyyyyyy] Error 1
```

I'm working under Ubuntu 14.04 and found through [this stackoverflow thread](http://stackoverflow.com/questions/15608236/eclipse-and-google-app-engine-importerror-no-module-named-sysconfigdata-nd-u) that this can be addressed one of two ways. However, updating `virtualenv` did not solve the issue, forcing me to run the following:

```
sudo ln -s /usr/lib/python2.7/plat-*/_sysconfigdata_nd.py /usr/lib/python2.7/
```

Is there a way we can solve this without having to force a user to run some kind of external command?

@commodo 
@hnyman

commodo, 2015-12-17:
hmm
will take a look tomorrow

dfc-d, 2015-12-17:
Must be an Ubuntu-specific issue.  On CentOS I have no such issue.

Regards,

Daniel

commodo, 2015-12-19:
I managed to build Python 2.7.11 on a Ubuntu 14.04 (x64) VM.4
I managed to build it also for target `ar71xx` and `imx6`
Any more details at this point wold help.

* Did you try this with a full clean build ? (i.e. make dirclean then build again ?)
* Are you building from trunk with the latest everything ?
* Is your Ubuntu 14.04 system 32 or 64 bit ?
* What's the target you're building (imx6, bcm2709) ?
* Is this OpenWRT core you're building or some custom OpenWRT derivative ? (your company specific, or some custom OpenWRT derivatives like Gateworks)
* Some more log console output would help.

psidhu, 2015-12-21:
To you bullet points: Yes, clean build. Latest trunk. 64bit. imx6 target, but can reproduce for any.  Using standard OpenWrt (not company specific). Extra log: http://pastebin.com/ii9ci8tr

I installed Ubuntu 14.04.3 LTS in a VM and found that the `_sysconfigdata.py` existed in the `/usr/lib/python2.7/` (same as you). I believe my specific problem is I had 13.04 LTS installed on my host, but at some point in the past ran an update to bump to 14.04 which never addressed the [actual bug](https://bugs.launchpad.net/ubuntu/+source/python2.7/+bug/1115466). However, in current releases of Ubuntu, it seems that the problem has been addressed.

This probably isn't an issue we should solve in the OpenWrt python Makefile, but should maybe be documented for anyone else that might run into this. This 'issue' on github might be enough for people to google around for the error and find a solution.

commodo, 2016-01-04:
@psidhu 
Thanks for identifying the issue :)

Any other reasons against closing this ?

Thanks

psidhu, 2016-01-05:
Nope, I'll close it out.

closed 2016-01-05 by psidhu