openwrt/packages #2152: seafile does not properly cross compile (builds various binaries against host arch instead of target)

dfc-d, 2015-12-20:
I'm going to do a series of issue reports with seafile in case someone cares to fix the issues which led to my issuing a PR to move seafile to packages-abandonded.

For this issue:

seafile builds a number of binaries for the host arch instead of target arch, when host arch does not match target arch (i.e. x86 cross-compiling for ar71xx).

This occurs for both seafile-server and seafile-seahub because of the way in which the Python builds are invoked.

For seafile-seahub the primary offender I am aware of PIL which is pulled in through easy_install of Djblets and would actually require patches (on PIL) to properly cross-compile.

kissg1988, 2017-10-07:
This has been fixed by properly packaging all the python dependencies.

See https://github.com/openwrt/packages/pull/2651

cshoredaniel, 2018-07-09:
This should be closed

thess, 2018-07-09:
OK by me

closed 2018-07-09 by thess