openwrt/packages #2153: seafile downloads and installs python dependencies during build instead of doing proper packaging

dfc-d, 2015-12-20:
I'm going to do a series of issue reports with seafile in case someone cares to fix the issues which led to my issuing a PR to move seafile to packages-abandonded.

For this issue:

make download (when building as part of buildroot) should download everything needed to compile a package at which point you should be able put the downloaded source files in a directory and use CONFIG_LOCAL_MIRROR=file::///path/to/files to avoid needing internet access to rebuild after e.g. distclean and reconfigure.

This fails with seafile because seafile employs a squirrelly download dependencies as part of the build process; but fails to actually download all dependencies (e.g. misses PIL which Djblets pulls in as part of easy_install because it's *require*d in Djblets setup.py).

In addition a properly packaged package should have shared code (e.g. python modules useful to other python programs) as packages which can be DEPENDed on by other packages (and of the course the one in question).

dfc-d, 2015-12-20:
Further if the issue is a need for specific python package versions, the package should be adding virtualenv and setting up a virtualenv for the non-shared environment.

kissg1988, 2017-10-04:
This has been fixed in https://github.com/openwrt/packages/pull/2651, please someone close the issue. Thank you!

closed 2018-08-18 by dibdot