openwrt/packages #2155: xl2tpd/mwan3 issues with 2 or more connections

psyborg55, 2015-12-20:
running r47929 with mwan3 configured to map main wan connection and 2 l2tp connections to different virtual wifi interfaces and experiencing the following problems:

upon boot both connections are established properly but i can only ping gateway of the first l2tp server (10.10.1.1) when connected to wifi interface that forwards to the first l2tp server.

if i connect to the 2nd wifi interface i can't ping gateway of 2nd l2tp connection and can't access services running there (ssh, rutorrent for example). but if i stop both connections and dial only the 2nd connection then the gateway of 2nd l2tp server is accessible. dialing first l2tp connection now results in first l2tp server inaccessible.

even more important the connection isn't dialed through wan gateway but through 2nd l2tp connection resulting in performance slowdown. this bug occurs whenever one of the connections goes down and reconnects. last time i used this kind of setup on 12.09 there were no such issues.

note that internet works on both interfaces regardless of the routes through which connections are established, of course with slowdown as i already mentioned when one tunnel is established through another instead through wan.

Adze1502, 2015-12-24:
Could you create a forum post on openwrt with the usual troubleshooting data. I will take a look as i think it is a configuration issue. Some things have changed in mwan3 version 2.0

psyborg55, 2016-01-01:
here you go: https://forum.openwrt.org/viewtopic.php?pid=305803#p305803

Adze1502, 2016-01-04:
My guts say this is a general problem and not an mwan3 specific problem, as i suspect that you also cant ping gateway of 2nd l2tp connection if mwan3 is not installed...

psyborg55, 2016-01-09:
tried fresh trunk build with 2.0-2 version, no improvement.

psyborg55, 2016-01-11:
i also think it's a general problem. today i've been using only one l2tp server (the other one has some connection problem and i can't restart it untill some download there finishes) and still i was unable to ssh into this server or access rutorrent/rapidleech.

now this happens after reboot with default configuration to auto-start both l2tp connections, but because of problems with 2nd server this connection isn't established. as soon as i click connect in luci for the 2nd connection, even though it's not connected, services on the first server are immediately available.

this must be netifd/routing/firewall problem. 

i haven't noticed this before with 12.09 because:

1. i was using PIA vpn and obviously had no services running on gateways 
2. all l2tp connections shared same firewall rule but i found it's better to separate them

Adze1502, 2016-01-12:
Please close this issue as it is not a mwan3 package error, but rather a general error. Create a OpenWRT trac ticket if necessary.

closed 2016-01-12 by psyborg55