openwrt/packages #2167: Segfault in Openconnect libc.so

pynu, 2015-12-23:
Firmware Version OpenWrt Designated Driver r47928 / LuCI Master (git-15.351.05963-967bb1f)
Kernel Version 3.18.23
Platform X86, X86-64

Thu Dec 17 16:40:10 2015 kern.info kernel: [ 129.910423] ocserv[3831]: segfault at 0 ip 00007f3ab235533b sp 00007fff3c0045d0 error 4 in libc.so[7f3ab233a000+66000]
Thu Dec 17 16:40:15 2015 kern.info kernel: [ 134.913907] ocserv[3842]: segfault at 0 ip 00007f776c78633b sp 00007ffee05fb630 error 4 in libc.so[7f776c76b000+66000]
Thu Dec 17 16:40:20 2015 kern.info kernel: [ 139.917354] ocserv[3853]: segfault at 0 ip 00007fdb032e633b sp 00007ffcf91294f0 error 4 in libc.so[7fdb032cb000+66000]
Thu Dec 17 16:40:25 2015 daemon.info procd: Instance ocserv::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Thu Dec 17 16:40:25 2015 kern.info kernel: [ 144.920830] ocserv[3864]: segfault at 0 ip 00007f24ee85b33b sp 00007ffdf69a05a0 error 4 in libc.so[7f24ee840000+66000]
Thu Dec 17 16:41:04 2015 kern.info kernel: [ 183.177266] ocserv[4248]: segfault at 0 ip 00007faaddac033b sp 00007fff9339d050 error 4 in libc.so[7faaddaa5000+66000]

t0t3m1n, 2016-01-05:
Hi, i have a similar problem with a more recent fw version

Firmware version: OpenWrt Designated Driver r48078 / Luci  Master (git-15.365.40842-0f1e39a)
Kernel version: 4.1.13
Platform: mips (Lantiq)

`Tue Jan  5 18:31:19 2016 kern.info kernel: [11487.676862] 
Tue Jan  5 18:31:19 2016 kern.info kernel: [11487.676862] do_page_fault(): sending SIGSEGV to ocserv for invalid read access from 00000000
Tue Jan  5 18:31:19 2016 kern.info kernel: [11487.683883] epc = 774dfb2c in libc.so[774c8000+92000]
Tue Jan  5 18:31:19 2016 kern.info kernel: [11487.688943] ra  = 00413280 in ocserv[400000+55000]
Tue Jan  5 18:31:19 2016 kern.info kernel: [11487.693562] 
Tue Jan  5 18:31:19 2016 daemon.info procd: Instance ocserv::instance1 s in a crash loop 6 crashes, 0 seconds since last crash`

nmav, 2016-01-05:
I do not seem to be able to reproduce that. Could you send me a generated configuration file that crashes? It should be in /tmp/etc/ocserv.conf

nmav, 2016-01-05:
Never mind, I think I figured the issue. I'll submit a fix and I'd appreciate testing.

t0t3m1n, 2016-01-06:
I've tested your patch and i can confirm it's working, no more segmentation fault with expose-iroutes = true. Thank you.

closed 2016-01-06 by nmav