openwrt/packages #2170: mpd cannot view tags in mp3 files

neheb, 2015-12-24:
FLAC and MP4 work. Just not mp3. Worked before in Chaos Calmer.

thess, 2015-12-24:
mpd-full or mpd-mini? Were you using ffmpeg - which version?

neheb, 2015-12-24:
mpd-mini. i can't seem to find mpd-full in make menuconfig for some reason. opkg also doesn't list it.

also, i don't have ffmpeg installed. is that the issue?

thess, 2015-12-24:
mpd-full only shows up if one of the ffmpeg libraries are selected. ffmpeg-audio-dec includes all audio decoders. mpd-mini should do the job for you without ffmpeg -- I'll look into it.

thess, 2015-12-29:
I have found no problems using mp3, flac & ogg format files with mpd-mini. mpd-mini does not support m4a/aac/mp4. There is one thing though if you are building your own version - be sure to have BUILD_PATENTED enabled to include mp3 support. Could that be your problem?

neheb, 2015-12-29:
Argh I switched to mpd-full so i can't test this again. Anyway, the issue was just tags. Playback of mp3 files worked fine. Patended functionality was enabled on my build. Well anyway, i'll close this since mpd-full fixes it.

closed 2015-12-29 by neheb