openwrt/packages #2193: [patch] ZNC FTBFS due to ICU dep

bastien-roucaries, 2015-12-27:
znc is FTBFS due to libicu not found.

using ICU is not an option due to huge size.

Better is to disable ICU by passing --disable-charset as configure flag.

Bastien

srdjanrosic, 2015-12-31:
PKG_MAINTAINER:=Jonas Gorski <jogo@openwrt.org>
  aka.
KanjiMonster

KanjiMonster, 2016-01-08:
Sorry for the late fix, was knocked out by a cold. In the future, please prefix my nick with @ so I get a notification for being mentioned.

closed 2016-01-08 by KanjiMonster

bastien-roucaries, 2016-01-09:
@KanjiMonster Still FTBFS see http://buildbot.openwrt.org:8010/broken_packages/ar71xx/znc/compile.txt should you do something on trunk (refresh ?)

KanjiMonster, 2016-01-09:
I pushed an update to 1.6.2 at the same time. The build failure is with 1.6.1, so this build was started before the fix. You'll need to wait until the next build iteration.

commit b1a2b9d1 (2016-01-08): znc: enable features/support explicitly

Don't let znc autodetect library presence to avoid random missing
dependencies popping up.

Closes #2193.