openwrt/packages #2201: [Discussion] OpenVSwitch upgrade to version 2.5.0

commodo, 2015-12-29:
Taken from this thread
http://comments.gmane.org/gmane.comp.embedded.openwrt.devel/37336

commodo, 2015-12-29:
Ah, I just noticed this PR #2197

commodo, 2015-12-29:
Moving on (in reply to the email thread)

The LTS is pretty much out of the question, since it supports up to kernel 3.14 ; and backporting support for it (even if doable) it's not much worth it.
So, adding a new package for the OVS LTS is a no go.
The current version in the feeds is 2.4.0 had support for 4.1 backported to it.

One remark is : I did add version 2.4.0 before it was released and I recall having some not-fun experiences, because it was a trunk version and stuff still needed to be fixed.
Also, OVS gets bloatier with each new version (more RAM, and more flash usage).

Having said that, let's add version 2.5.0, since it seems the request for the most bleeding OVS is a repeating one.
I will try to handle whatever fallout it comes after that.
Maybe some people are doing research with SDN and using OpenWRT with OVS in some combinations.

This week I was trying to split OVS into smaller packages and add --disable configure flags.
So this should handle the bloat aspect, by allowing stuff to not be compiled or to package it separately.

jbemmel, 2015-12-29:
Just created a new clean pr for your convenience: https://github.com/openwrt/packages/pull/2202

I need 2.5.0 not because it's the latest OVS, but because my ramips target device uses Linux kernel 4.3 and 2.4.0 only supports up to kernel 4.1

It would be nice to have flags to trim down the package, for example STT and GRE tunneling support may not be needed in all cases.

commodo, 2015-12-29:
Hmmm,

allow me to correct one of my previous affirmations that seems misleading/erroneous:
> Having said that, let's add version 2.5.0, since it seems the request for the most bleeding OVS is a repeating one.

Actually the requests that I got were more that OVS to run on OpenWRT, and since OpenWRT is shifting kernel support faster than OVS, it needs a more recent (even bleeding edge) of OVS.

A couple of months ago, there were 2 requests to support OVS on OpenWRT with kernel 4.1.
And that's approximately when it was added in.

Sorry for the confusion.

closed 2015-12-30 by commodo