openwrt/packages #2208: Openwrt on Panic 222HD DVB S2 receiver

segabits, 2015-12-30:
hi everyone,
I just brought a very cheap DVB S2 receiver "http://www.aliexpress.com/item/HD-Mini-FTA-Digital-Satellite-Receiver-DVB-S-S2-Full-HD-MPEG-2-MPEG-4-H/32455745445.html", and the video quality is very bad, shuttering on HD channel and very bad upscaling on the SD one, I was going to throw it away but I figure out that the box is very very compact has a DVB S2 TUNER and USB 2.0 port, if we can run OPENWRT on it and with TVHEADEND it will become a killer cheap device to stream video to other TV boxes.
the device run an ALI3510 chipset @ 400 Mhz (MIPS), 128 MB of RAM and a ROM of 8 MB really ideal for OPENWRT.
the idea a wanted to extract the BOOTLOADER and KERNEL from the firmware and add to it an OPENWRT ROOTFS with modules for DVB s2 TUNER and a USB DONGLE
the idea seems boald and I don't now if it would work at first place
any opinion HELP or new idea how to proceed will be welcome 
thanks

plntyk, 2016-01-06:
This is the wrong place for your question. 
If you want to port : please ask for help in [OpenWrt forums](forum.openwrt.org).

PS: You project success depends if you have serial console, drivers for that DVB chip in Linux Kernel - even then you might have to do some C coding yourself.

thess, 2016-01-10:
As @plntyk said - this item does not belong here. Asking for support for your project in the forums would be a better route to go.

closed 2016-01-10 by thess