openwrt/packages #2244: gcc fails to build [buildbot,replicated]

plntyk, 2016-01-06:
gcc is failing on several targets:

- [ar71xx](http://buildbot.openwrt.org:8010/broken_packages/ar71xx/gcc/compile.txt)
- [brcm47xx](http://buildbot.openwrt.org:8010/broken_packages/brcm47xx/gcc/compile.txt)
- [x86](http://buildbot.openwrt.org:8010/broken_packages/x86/gcc/compile.txt)

```
Package gcc is missing dependencies for the following libraries:
libc.so.6
libm.so.6
```
This dependency looks wrong (other musl binaries use "libc.so")

Culprit is **libcc1.so.0.0.0**
It seems wrong that this library is copied from the OpenWrt toolchain dir of GCC5 and then used in GCC 4.8.3. 

another issue: --disable-libsanitizer might be useful

plntyk, 2016-01-06:
libcc1 is not part of GCC 4.8.3. So copying it is obviously wrong.

courtjestr, 2016-01-10:
If you change the GCC compiler Version in the toolchain options from gcc 5.x to gcc 4.8.x with Linaro enhancements then gcc builds successfully.

jow-, 2016-01-12:
@bk138

dmamalis, 2016-01-19:
I have successfully built the default configuration for the ar71xx, however a custom configuration of mine does fail with the same dependency problem. Since this is a gcc dependency how does it not come up during the default configuration built?

xypron, 2016-01-23:
Package GCC should always compile the same version of GCC as the toolchain to ensure ABI compatibility. For instance the C++ ABI of gcc 5.2 is incompatible to gcc 4.8.3.

Hence in the Makefile we have to evaluate GCC_VERSION to decide which version of GCC we compile. See toolchain/gcc/common.mk.

Furthermore we should ensure that we use the same flags. E.g. if for the ARM architecture we compile the toolchain with --hard-float we should do so in package GCC too (cf. common.mk).

bk138, 2016-01-31:
Hey, are we talking about different issues here? I can't build gcc because another package, namely 'toolchain' fails to build libgcc:
`cp: cannot stat '/home/bk/Projekte/PROGGA/00shoutr/openwrt-trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.12/lib/libgcc_s.so.*': No such file or directory
`

KamathKrish, 2016-04-07:
I am hitting the same error. Is this problem unique to me or others still seeing it? This is on openwrt trunk from this week.

`Package gcc is missing dependencies for the following libraries:
libc.so.6
`

ppelleti, 2016-04-08:
@KamathKrish, yes, I'm seeing:

```
Package gcc is missing dependencies for the following libraries:
libc.so.6
```

after I add `--disable-libsanitizer` to `feeds/packages/devel/gcc/Makefile`.  (Otherwise, I run into #2581 first.)

ppelleti, 2016-04-09:
@courtjestr, how do I change the toolchain compiler version to 4.8.x?  When I go to Advanced Configuration Options > Toolchain Options > GCC compiler Version, there is only a single option "gcc 5.x".  How do I get 4.8.x to appear as a choice there?

bk138, 2016-04-15:
Finally, I can reproduce this myself as well. (--disable-libsanitizer is needed, btw)

@plntyk Yes, copying files over from the toolchain dir is a gross hack and prone to break stuff. As we can see now. This is from the old Makefile from oldpackages and - unfortunately - I can't see an easy way around this as, AFAICT, those crt*.o files are needed by gcc. They do *not* get installed in  $(PKG_INSTALL_DIR) though. Seems to me that the cleanest solution would be to package libgcc as well. 

Anyway, I am not using on-target gcc anymore, so am willing to hand over maintenership to someone more intrinsically motivated. If noone steps up, it's okay for me to continue as maintainer, but I definitely need some help with this issue here.

zyxmon, 2016-04-18:
FYI I compile gcc 5.3.0 package (based on openwrt Makefile) in Entware for arm, mipsel, i386, x86_64 - 
https://github.com/Entware-ng/entware-packages/tree/master/devel/gcc

rkoornstra, 2016-04-26:
I'm using CONFIG_GCC_VERSION="5.3.0"
The default library is now musl.
However, during the build process I see:

Package gcc is missing dependencies for the following libraries:
libc.so.6

It seems some patches are needed to apply to gcc before compiling gcc?
Where is a good place to put patches in the build tree to be applied before compiling gcc?
Thanks,

Reinoud

bk138, 2016-04-26:
@rkoornstra Well, the patches dir ;-)

rkoornstra, 2016-04-27:
Reading the release notes for GCC 6.1, MUSL is finally supported.
It can be selected using the new -mmusl option. :)

Another question: First the cross tools are being created, including a cross gcc.
After that we don't need to build gcc again, unless you want gcc on the target system.
Is this an option I can disable?

noblepepper, 2016-05-18:
I have a branch that compiles and works fine on mipsel targets at https://github.com/noblepepper/packages/tree/updategccTo5_3

I am working on general clean up and fixing some of the issues raised in this thread but if anyone can test (and maybe fix problems) on other targets I would be very interested in the results. 

Hopefully we can come up with something deserving of a pull request to this repository!

Thanks

bk138, 2016-05-26:
Please notify me when you're ready @noblepepper. Thanks!

rmanders, 2016-07-26:
I'm having a similar issue while trying to compile for the Raspberry Pi3 profile: 

- **Target System**: `Broadcom BCM27xx`
- **Subtarget**: `BCM2710 based boards`
- **OpenWrt trunk commit**: `558b38f4b8e615fb5262c94dad7db2071e394ac1`
- **Package feeds commit**: `fb37ae282b4d3dbbd1c1a92b1747a6806b182493`

When compiling with the GCC package I get the error:

```
Package gcc is missing dependencies for the following libraries:
libc.so.6
```

If there are any ideas on how to work on a fix for this, I'd be happy to test/try it

paulo-raca, 2016-08-12:
Same issue here.
Is there a workaround?

mozzwald, 2016-08-12:
@paulo-raca We are using this patch for [openwrt-zipit](https://github.com/openwrt-zipit/) (arm/pxa Zipit Z2 Handheld) based off @noblepepper's branch. It works for us.

https://github.com/openwrt-zipit/openwrt-pxa-target/blob/master/patches/openwrt-packages/003-gcc-bump-to-5.3.0.patch

bk138, 2016-08-17:
@mozzwald care to file a PR so we can integrate this here?

mozzwald, 2016-08-17:
@bk138 I'm not sure the patch is the correct fix for upstream openwrt-packages. It would not work for boards using other than gcc 5.3. [xypron's comment](https://github.com/openwrt/packages/issues/2244#issuecomment-174228430) above is likely the best route to take and I dunno how to go about that in the openwrt Makefile.

noblepepper, 2016-09-25:
@bk138 Apologies for the long silence, I've been off in Chaos Calmer world. 

I can't dispute xypron's comment about abi's since I ran into compile problems if the package and toolchain don't match so I made them match.

I have tested on mt7688 & mt7628 which are mipsel, while I have only tested "hello world" on my test images similar images have built an entire AVR toolchain.

It does compile for arm, I picked Broadcom BCM47xx/53xx (ARM) at random, but I don't have hardware to do further testing.

bk138, 2016-09-26:
@noblepepper thank you very much for this! Care to take over maintainership of this package? As I mentioned earlier, I am not actively using on-target gcc anymore, so someone who actually needs that package might be suited better for the task.

noblepepper, 2016-09-27:
I'm not sure how much time will be needed or I can spare but since you aren't using this package anymore I can give "maintainership" a try. Let me know how we go about doing this.

bk138, 2016-09-27:
cool. you just set yourself as maintainer in the makefile. that's it ;-)

noblepepper, 2016-09-28:
Done. Is there any thing that needs to be done for handling pull requests, etc?

bk138, 2016-09-28:
well lede/openwrt guys will cc you when a pr comes in. you bring in the technical expertise, but as they are owners of the tree, they will have the last word. that's about it.

hnyman, 2016-10-04:
gcc has been updated to 5.4.0 and seems to compile ok in buildbots

closed 2016-10-04 by hnyman