openwrt/packages #2248: tor service don't start

artemklevtsov, 2016-01-07:
Hello.

Tor daemon can't start because `tor` user does not exists.

```text
:~# tor
Jan 07 00:28:19.907 [notice] Tor v0.2.7.6 running on Linux with Libevent 2.0.22-stable, OpenSSL 1.0.2e and Zlib 1.2.8.
Jan 07 00:28:19.908 [notice] Tor can't help you if you use it wrong! Learn how to be safe at https://www.torproject.org/download/download#warning
Jan 07 00:28:19.909 [notice] Read configuration file "/etc/tor/torrc".
Jan 07 00:28:19.961 [notice] You configured a non-loopback address '192.168.1.1:9100' for SocksPort. This allows everybody on your local network to use your machine as a proxy. Make sure this is what you wanted.
Jan 07 00:28:19.972 [notice] You configured a non-loopback address '192.168.1.1:9100' for SocksPort. This allows everybody on your local network to use your machine as a proxy. Make sure this is what you wanted.
Jan 07 00:28:19.973 [notice] Opening Socks listener on 192.168.1.1:9100
Jan 07 00:28:19.974 [warn] Error setting configured user: tor not found
Jan 07 00:28:19.976 [warn] Failed to parse/validate config: Problem with User value. See logs for details.
Jan 07 00:28:19.976 [err] Reading config failed--see warnings above.
```

karlp, 2016-05-24:
what branch is this on?  Are you still seeing this?  I see that the trunk tor makefile includes user ids to be made, and has done since oct 2014, even in the 15.05 branch

hauke, 2017-01-14:
I am unable to reproduce this with LEDE and current packages feed.

closed 2017-01-14 by hauke