openwrt/packages #2250: Adblock filtering issue

ASAPHAANING, 2016-01-07:
the openwrt package for Designated Driver (trunk of 15) is not filtering ads on YouTube

dibdot, 2016-01-09:
Sorry, I have no crystal ball ... 
- which adblock sources are enabled in config?
- example link (with ads)?
Thanks!

ASAPHAANING, 2016-01-09:
I'll interpret that as failure to include specifications that can lead to effective troubleshooting, am I correct?

I'll try to elaborate on your points and hopefully beyond:

The only ads that escape the filter, which currently has the following sourcefiles enabled:
adaway
disconnect
shalla (all)
yoyo
whocares (disabled for now though, tested youtube with this enabled as well)
are youtube ads on a mobile device (iOS9) which I assume is https.

Link is a little troublesome since I can't guarantee an ad in a specific video, but applies to basically all videos that have not been cached by the device. 

Apps in particular do not suffer from this issue in general, and I've confirmed multiple apps with ads ridden effectively.

dibdot, 2016-01-09:
I can't reproduce this issue ... maybe you are interested in testing my current development release? Please drop me a short mail (see maintainer email). Thanks!

ASAPHAANING, 2016-01-09:
To whom it may concern: 

Dibdot provided a solution to my situation by adding the following firewall DNS rule to /etc/firewall.user:

# redirect/force all dns queries to port 53 of your router
# configuration found in /etc/firewall.user
#
iptables -t nat -I PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -I PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53

After that, please restart your firewall (/etc/init.d/firewall restart) and try again

closed 2016-01-09 by ASAPHAANING