openwrt/packages #2254: Auto-updating OpenWrt packages on boot

anirudt, 2016-01-09:
This is a feature, I understand, for certain reasons, owing to the number of platforms it is utilized on. However, wouldn't it be prudent to make a package to add this package auto-updating feature on bootup? I would be happy to know if this would be useful, and would take this up if required.

bittorf, 2016-01-09:
makes sense, but has a lot of issues to solve (e.g. aborted updates, lost connection, hanging wget). you can join the discussion on the mailinglist maybe: http://lists.prplfoundation.org/pipermail/openwrt/2016-January/000460.html

anirudt, 2016-01-09:
Sure. I've something in mind, I'll add the commits and maybe we can discuss over it.

bittorf, 2016-01-09:
@anirudt sounds good.

anirudt, 2016-01-10:
I do understand that it is an init version, but features could be added over a time, and this could be a good place to start from.

dfc-d, 2016-01-10:
Terribly bad idea.

Likely the single biggest issue is that packages which are in the 
squashfs will get updated with packages in the overlay, but the overlay 
is likely not big enough to handle all updated packages.

At the very least autoupdate should only update packages already on the 
overlay or which are on a normal r/w filesystem.

Also need to need recover gracefully from 'out-of-space' that can easily 
occur due to fact that due the compression method used there is no 
failsafe way to predict how much space a package will use on a given 
device (so can't avoid the possibility of exceeding storage on overlay).

Regards,

Daniel

anirudt, 2016-01-10:
Right, I understand. 

Not having a consolidated platform for testing right now, will focus on it as time proceeds.

closed 2016-01-10 by anirudt

dfc-d, 2016-01-12:
One thing that could help is if you modified opkg to abort/rollback if a 
write would cause there to be less than one (or better two) erase blocks 
work of jffs2 (assuming stock image on a router vs something like ext4 
elsewhere, where you probably still should worry about ending up with 
not enough usable space) left.

That way you would never break the router by making the filesystem 
unwriteable and IMO would be a good thing for openwrt to have anyway.

Regards,

Daniel