openwrt/packages #2256: NUT init script no longer creates ups.conf from UCI

RoganDawes, 2016-01-09:
Having upgraded from 14.07 to 15.05, I noticed that my NUT config was no longer working.

This appears to be because of changes to the init scripts for https://dev.openwrt.org/browser/branches/packages_14.07/utils/nut/files/ups.init. Here you can see the config file is reconstructed as /var/run/ups.conf in the start() function. The makefile for trunk now shows that the config file is expected to reside in /etc/nut/ups.conf, but there is no mechanism to create this file from the UCI data.

RoganDawes, 2016-01-09:
As can be seen in the makefile at https://github.com/openwrt/packages/blob/6981734402295519275c0d72c11924e7a146089b/net/nut/Makefile#L320, nut is looking for its configuration file in /etc/nut, which is not dynamically created (i.e. not /var/run)

cyanidium, 2016-01-18:
When porting the package from oldpackages I removed the UCI option due to lack of documentation and additional complexity. As stated in https://github.com/openwrt/packages/pull/1803#issuecomment-147019156, I am happy to merge and support a pull request that re-implements the functionality, I just think that it should be documented appropriately. Without the documentation, I would rather stick to the upstream supported config mechanism. Since this is easily supported by the default overlayfs setup, I don't see the need for the additional complexity of creating the config file from UCI, and especially not until a UCI implementation is at feature parity with the upstream method.

UCI (and LUCI) support is on my list of things to do, but very far down. In the meantime this is a "wontfix" issue.

RoganDawes, 2016-01-18:
Understood, I'm happy with that answer.

closed 2016-01-18 by RoganDawes