openwrt/packages #2259: ncdu dependency

skonvols2k, 2016-01-10:
Hi,
dependencies of package ncdu are wrong, in fact it needs terminfo package, but isn't installed for dependency.
if package terminfo isn't present ncdu won't start with error:
Error opening terminal: ansi.
I'm using CHAOS CALMER (15.05, r46767) on ar71xx generic.
Regards

hnyman, 2016-01-12:
cc to ncdu maintainer: @clehner

clehner, 2016-01-12:
ncdu depends on libncurses, and libncurses depends on terminfo. Is this not right?

skonvols2k, 2016-01-12:
Hi,
this is what results to me:

root@TL-WDR4300:~# opkg depends ncdu
ncdu depends on:
        libc
        libncursesw
root@TL-WDR4300:~# opkg depends libncursesw
libncursesw depends on:
        libc

Regards

---------------------------------------
Marco Sommella
marco.sommella@gmail.com
---------------------------------------

hnyman, 2016-01-12:
In CC15.05 that @skonvols2k uses, ncdu is still using the "old" Makefile before the last version bump and change from libncursesw to libncurses
CC15.05: https://github.com/openwrt/packages/blob/for-15.05/utils/ncdu/Makefile
trunk:  https://github.com/openwrt/packages/blob/master/utils/ncdu/Makefile

Additionally, libncursesw has been broken to some extent if compiled without libncurses being also present in the system. That was recently discussed at mutt thread: https://github.com/openwrt/packages/pull/1997#issuecomment-162656160 and then fixed (at least to some extent) in trunk with https://dev.openwrt.org/changeset/47853 

The version bump to 1.11 and change to using libncurses might maybe be backported to CC15.05.

clehner, 2016-01-12:
@hnyman thanks for explaining that. I will open a PR to backport the changes to CC15.05.

hnyman, 2016-01-12:
#2262 should have fixed this in CC15.05 if ncdu is self-compiled.

PR #2262 (2016-01-12): ncdu: update to 1.11
This fixes #2259 where terminfo was not getting pulled in as a dependency of ncdu via libncurses, by applying changes from master to for-15.05.
closed 2016-01-12 by hnyman