openwrt/packages #2268: dynamic_dns_functions get_local_ip() filters /128 adress

karlsebal, 2016-01-15:
get_local_ip() line 854 filters /128 adress, but the only global address I get from my modem is with that bitmask. This address is working and ping6able. This behaviour may be intended but as for me its a bug.

chris5560, 2016-01-15:
@karlsebal
If I understand you right the /128 is the only addr you get from your provider.
This addr is pingable.
This addr is NOT handled by ddns_scripts if you use "interface" as ip_source.

line 854 should only show addresse NOT fxxx and NOT /128
please send me your ifconfig output of your wan6 interface so I can have a look on it.

karlsebal, 2016-01-22:
Yes, that's right, the /128 address is the one I get from my provider 
and the one I wanna update via ddns-scripts for it's the working one. So 
I do not understand clearly why "! \128" is intended. My ifconfig shows

eth0.2    Link encap:Ethernet  HWaddr C0:4A:00:38:EB:EA
           inet addr:92.206.35.7  Bcast:92.206.47.255 Mask:255.255.240.0
           inet6 addr: fe80::c24a:ff:fe38:ebea/64 Scope:Link
           inet6 addr: 2a00:c1a0:c000:0:6832:7c9b:c61e:2785/128 Scope:Global
           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
           RX packets:15237278 errors:0 dropped:0 overruns:0 frame:0
           TX packets:3651431 errors:0 dropped:0 overruns:0 carrier:0
           collisions:0 txqueuelen:0
           RX bytes:8936825322 (8.3 GiB)  TX bytes:882633340 (841.7 MiB)

Thank you :)
Karl

chris5560, 2016-01-24:
Will be fixed in next update
Christian

chris5560, 2016-01-26:
@karlsebal
Fixed in ddns-scripts 2.6.0-1 merged yesterday.
Could you please close.
Thanks
Christian

karlsebal, 2016-01-27:
Thank you very much!

Karl

closed 2016-01-27 by karlsebal