openwrt/packages #2291: Squid compile error

taeasy, 2016-01-19:
lrt -ldl -ldl -L.. -lltdl 
libtool: link: rm -f .libs/squid.nm .libs/squid.nmS .libs/squid.nmT
libtool: link: (cd .libs && i486-openwrt-linux-uclibc-gcc -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -c -fno-builtin "squidS.c")
libtool: link: rm -f ".libs/squidS.c" ".libs/squid.nm" ".libs/squid.nmS" ".libs/squid.nmT"
libtool: link: i486-openwrt-linux-uclibc-g++ -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Werror -pipe -D_REENTRANT -I/home/main/CC_x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -std=c++11 .libs/squidS.o -o squid AclRegs.o AuthReg.o AccessLogEntry.o AsyncEngine.o YesNoNone.o cache_cf.o CacheDigest.o cache_manager.o carp.o cbdata.o ChunkedCodingParser.o client_db.o client_side.o client_side_reply.o client_side_request.o BodyPipe.o clientStream.o CollapsedForwarding.o CompletionDispatcher.o ConfigOption.o ConfigParser.o CpuAffinity.o CpuAffinityMap.o CpuAffinitySet.o debug.o delay_pools.o DelayId.o DelayBucket.o DelayConfig.o DelayPool.o DelaySpec.o DelayTagged.o DelayUser.o DelayVector.o NullDelayId.o ClientDelayConfig.o disk.o DiskIO/DiskIOModule.o DiskIO/ReadRequest.o DiskIO/WriteRequest.o dlink.o dns_internal.o DnsLookupDetails.o errorpage.o ETag.o event.o EventLoop.o external_acl.o ExternalACLEntry.o FadingCounter.o fatal.o fd.o fde.o filemap.o fqdncache.o FwdState.o gopher.o helper.o htcp.o http.o HttpHdrCc.o HttpHdrRange.o HttpHdrSc.o HttpHdrScTarget.o HttpHdrContRange.o HttpHeader.o HttpHeaderTools.o HttpBody.o HttpMsg.o HttpParser.o HttpReply.o RequestFlags.o HttpRequest.o HttpRequestMethod.o icp_v2.o icp_v3.o int.o internal.o ipc.o ipcache.o SquidList.o main.o MasterXaction.o mem.o mem_node.o MemBuf.o MemObject.o mime.o mime_header.o multicast.o neighbors.o Notes.o Packer.o Parsing.o pconn.o peer_digest.o peer_proxy_negotiate_auth.o peer_select.o peer_sourcehash.o peer_userhash.o PeerPoolMgr.o redirect.o refresh.o RemovalPolicy.o send-announce.o MemBlob.o SBuf.o SBufExceptions.o SBufDetailedStats.o SBufStatsAction.o SquidMath.o SquidNew.o stat.o StatCounters.o StatHist.o String.o StrList.o stmem.o store.o StoreFileSystem.o store_io.o StoreIOState.o store_client.o store_digest.o store_dir.o store_key_md5.o store_log.o store_rebuild.o store_swapin.o store_swapmeta.o store_swapout.o StoreMetaUnpacker.o StoreMeta.o StoreMetaMD5.o StoreMetaSTD.o StoreMetaSTDLFS.o StoreMetaURL.o StoreMetaVary.o StoreStats.o StoreSwapLogData.o SwapDir.o Transients.o MemStore.o time.o tools.o tunnel.o url.o urn.o wccp.o wccp2.o whois.o wordlist.o LoadableModule.o LoadableModules.o DiskIO/DiskIOModules_gen.o err_type.o err_detail_type.o globals.o hier_code.o icp_opcode.o LogTags.o lookup_t.o repl_modules.o swap_log_op.o DiskIO/Blocking/BlockingDiskIOModule.o DiskIO/DiskDaemon/DiskDaemonDiskIOModule.o DiskIO/DiskThreads/DiskThreadsDiskIOModule.o DiskIO/IpcIo/IpcIoDiskIOModule.o DiskIO/Mmapped/MmappedDiskIOModule.o -Wl,--export-dynamic  -L/home/main/CC_x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/main/CC_x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/main/CC_x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/main/CC_x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib auth/.libs/libacls.a ident/.libs/libident.a acl/.libs/libacls.a acl/.libs/libstate.a auth/.libs/libauth.a libBlocking.a libDiskDaemon.a libDiskThreads.a libIpcIo.a libMmapped.a acl/.libs/libapi.a base/.libs/libbase.a ./.libs/libsquid.a ip/.libs/libip.a fs/.libs/libfs.a ssl/.libs/libsslsquid.a ssl/.libs/libsslutil.a ipc/.libs/libipc.a mgr/.libs/libmgr.a anyp/.libs/libanyp.a comm/.libs/libcomm.a eui/.libs/libeui.a helper/.libs/libhelper.a http/.libs/libsquid-http.a icmp/.libs/libicmp.a icmp/.libs/libicmp-core.a log/.libs/liblog.a format/.libs/libformat.a clients/.libs/libclients.a servers/.libs/libservers.a ftp/.libs/libftp.a repl/liblru.a -lpthread -lcrypt adaptation/.libs/libadaptation.a parser/.libs/libsquid-parser.a ../lib/.libs/libmisccontainers.a ../lib/.libs/libmiscencoding.a ../lib/.libs/libmiscutil.a -lssl -lcrypto -lgssapi_krb5 -lkrb5 -lk5crypto -lcom_err ../compat/.libs/libcompat-squid.a /home/main/CC_x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/i486-openwrt-linux-uclibc/lib/libstdc++.so -lm -lrt -ldl -L.. -lltdl -Wl,-rpath -Wl,/home/main/CC_x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/i486-openwrt-linux-uclibc/lib -Wl,-rpath -Wl,/home/main/CC_x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/i486-openwrt-linux-uclibc/lib
/home/main/CC_x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/i486-openwrt-linux-uclibc/4.8.3/../../../../i486-openwrt-linux-uclibc/bin/ld: cannot find -lcom_err
/home/main/CC_x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib/libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
collect2: error: ld returned 1 exit status
libtool: link: rm -f ".libs/squidS.o"
make[6]: *** [squid] Error 1
make[6]: Leaving directory `/home/main/CC_x86/build_dir/target-i386_i486_uClibc-0.9.33.2/squid-3.5.12/src'
make[5]: *** [install-recursive] Error 1
make[5]: Leaving directory `/home/main/CC_x86/build_dir/target-i386_i486_uClibc-0.9.33.2/squid-3.5.12/src'
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/main/CC_x86/build_dir/target-i386_i486_uClibc-0.9.33.2/squid-3.5.12/src'
make[3]: *** [install-recursive] Error 1
make[3]: Leaving directory `/home/main/CC_x86/build_dir/target-i386_i486_uClibc-0.9.33.2/squid-3.5.12'
make[2]: *** [/home/main/CC_x86/build_dir/target-i386_i486_uClibc-0.9.33.2/squid-3.5.12/.built] Error 2
make[2]: Leaving directory `/home/main/CC_x86/feeds/packages/net/squid'
make[1]: *** [package/feeds/packages/squid/compile] Error 2
make[1]: Leaving directory `/home/main/CC_x86'
make: *** [package/squid/compile]  Error 2
main@debian:~/CC_x86$

dangowrt, 2016-01-19:
Please try cherry-picking https://github.com/openwrt/packages/pull/2264 into your feeds/packages repo:
```
cd feeds/packages
wget -O- https://patch-diff.githubusercontent.com/raw/openwrt/packages/pull/2264.patch | git am
```

taeasy, 2016-01-20:
i have updated the #2264 patch but the problem still

ffainelli, 2016-01-22:
Could you search your staging_dir directory for libcom_err*.so and print the results?

plntyk, 2016-02-11:
disable krb5 - see #2371

or fix the krb5 package to copy that libcom_err instead of removing it
maybe copy all krb5 libs to usr/lib/krb5

yousong, 2016-11-09:
I am closing this now as the package is present on LEDE snapshot download

closed 2016-11-09 by yousong