openwrt/packages #2293: Unable to compile udev/mtd_probe: unknown type name 'uint8_t'

Anaerin, 2016-01-20:
Compiling x86 (using the x86_64 profile) in a brand-new Ubuntu VM, and I get the following error:
```
make[3]: Entering directory '/home/anaerin/openwrt/package/system/udev'
CFLAGS="-Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/usr/include -I/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/include -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/usr/include -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/usr/include -I/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/include -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/usr/include -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/include " LDFLAGS="-L/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/usr/lib -L/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/lib -L/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/usr/lib -L/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/lib -znow -zrelro " make -j1 -C /home/anaerin/openwrt/build_dir/target-x86_64_musl-1.1.11/udev-173/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/home/anaerin/openwrt/build_dir/target-x86_64_musl-1.1.11/udev-173'
make --no-print-directory all-recursive
Making all in .
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I.  -include ./config.h -I./libudev -DSYSCONFDIR=\""/etc"\" -DLIBEXECDIR=\""/lib/udev"\" -I/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/usr/include -I/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/include -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/usr/include -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/anaerin/openwrt/staging_dir/toolchain-x86_64_gcc-5.2.0_musl-1.1.11/include  -fvisibility=hidden -ffunction-sections -fdata-sections -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT extras/mtd_probe/extras_mtd_probe_mtd_probe-mtd_probe.o -MD -MP -MF extras/mtd_probe/.deps/extras_mtd_probe_mtd_probe-mtd_probe.Tpo -c -o extras/mtd_probe/extras_mtd_probe_mtd_probe-mtd_probe.o `test -f 'extras/mtd_probe/mtd_probe.c' || echo './'`extras/mtd_probe/mtd_probe.c
In file included from extras/mtd_probe/mtd_probe.c:19:0:
extras/mtd_probe/mtd_probe.h:24:2: error: unknown type name 'uint32_t'
  uint32_t reserved;
  ^
extras/mtd_probe/mtd_probe.h:25:2: error: unknown type name 'uint8_t'
  uint8_t data_status;
  ^
extras/mtd_probe/mtd_probe.h:26:2: error: unknown type name 'uint8_t'
  uint8_t block_status;
  ^
extras/mtd_probe/mtd_probe.h:27:2: error: unknown type name 'uint8_t'
  uint8_t lba_copy1[2];
  ^
extras/mtd_probe/mtd_probe.h:28:2: error: unknown type name 'uint8_t'
  uint8_t ecc2[3];
  ^
extras/mtd_probe/mtd_probe.h:29:2: error: unknown type name 'uint8_t'
  uint8_t lba_copy2[2];
  ^
extras/mtd_probe/mtd_probe.h:30:2: error: unknown type name 'uint8_t'
  uint8_t ecc1[3];
  ^
Makefile:1892: recipe for target 'extras/mtd_probe/extras_mtd_probe_mtd_probe-mtd_probe.o' failed
make[6]: *** [extras/mtd_probe/extras_mtd_probe_mtd_probe-mtd_probe.o] Error 1
Making all in libudev/docs
make[6]: Nothing to be done for 'all'.
Makefile:2342: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
Makefile:1131: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/anaerin/openwrt/build_dir/target-x86_64_musl-1.1.11/udev-173'
Makefile:157: recipe for target '/home/anaerin/openwrt/build_dir/target-x86_64_musl-1.1.11/udev-173/.built' failed
make[3]: *** [/home/anaerin/openwrt/build_dir/target-x86_64_musl-1.1.11/udev-173/.built] Error 2
make[3]: Leaving directory '/home/anaerin/openwrt/package/system/udev'
package/Makefile:191: recipe for target 'package/system/udev/compile' failed
make[2]: *** [package/system/udev/compile] Error 2
make[2]: Leaving directory '/home/anaerin/openwrt'
package/Makefile:188: recipe for target '/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/home/anaerin/openwrt/staging_dir/target-x86_64_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/anaerin/openwrt'
/home/anaerin/openwrt/include/toplevel.mk:191: recipe for target 'world' failed
make: *** [world] Error 2
```

vpeter4, 2016-01-25:
I add include in mtd_probe.c and error goes away.
    #include <stdint.h>

joilg, 2016-01-27:
$ cat package/system/udev/patches/0004-fix-unknown-type.patch
--- a/extras/mtd_probe/mtd_probe.h
+++ b/extras/mtd_probe/mtd_probe.h
@@ -18,6 +18,7 @@
  */

 #include <mtd/mtd-user.h>
+#include <stdint.h>

 /* Full oob structure as written on the flash */
 struct sm_oob {

yousong, 2016-08-13:
udev was part of base packages and it's now removed from both OpenWrt and LEDE repo, so I am going to close this.

closed 2016-08-13 by yousong