openwrt/packages #2297: Openwrt: Compilation error with --enable-redir

drkhosla, 2016-01-21:
I am trying to compile coova chilli 06th Jan, 2016 using openwrt r48251. I think it is long pending issue.

libtool: compile:  mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I/var/firmware/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/var/firmware/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/include -I/var/firmware/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/include -I/var/firmware/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include/fortify -I/var/firmware/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -O2 -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -MT options.lo -MD -MP -MF .deps/options.Tpo -c options.c  -fPIC -DPIC -o .libs/options.o
options.c: In function 'options_fromfd':
options.c:381:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_host.allocated)
                                                ^
options.c:383:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_path.allocated)
                                                ^
options.c:385:46: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_qs.allocated)
                                              ^
make[7]: *** [options.lo] Error 1
make[7]: Leaving directory `/var/firmware/openwrt/build_dir/target-mips_34kc_musl-1.1.11/coova-chilli-1.3.0+20160106/src'
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory `/var/firmware/openwrt/build_dir/target-mips_34kc_musl-1.1.11/coova-chilli-1.3.0+20160106/src'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/var/firmware/openwrt/build_dir/target-mips_34kc_musl-1.1.11/coova-chilli-1.3.0+20160106'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/var/firmware/openwrt/build_dir/target-mips_34kc_musl-1.1.11/coova-chilli-1.3.0+20160106'
make[3]: *** [/var/firmware/openwrt/build_dir/target-mips_34kc_musl-1.1.11/coova-chilli-1.3.0+20160106/.built] Error 2
make[3]: Leaving directory `/var/firmware/drkhosla/coova-chilli'
make[2]: *** [package/feeds/drkhosla/coova-chilli/compile] Error 2
make[2]: Leaving directory `/var/firmware/openwrt'
make[1]: *** [/var/firmware/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory `/var/firmware/openwrt'
make: *** [world] Error 2

drkhosla, 2016-10-13:
Actually musl library re_pattern_buffer does not include allocated variable. https://fossies.org/dox/musl-1.1.15/structre__pattern__buffer.html

diizzyy, 2017-06-09:
@thess @hnyman 
Please close, not a pr.

@drkhosla 
If you're still interested, please use the forums (LEDE or OpenWrt) instead.

teslamint, 2019-03-27:
Since #5605 is closed, this issue can close now.

closed 2019-03-27 by neheb