openwrt/packages #2299: Strongswan 5.3.5, netlink error: No such file or directory (2)

pynu, 2016-01-22:
OpenWrt Designated Driver r48455 / LuCI Master (git-16.020.59380-63d70da) 
Kernel:4.4.0
Platform:X86_64, X86
Strongswan 5.3.5

Sat Jan 23 00:49:59 2016 daemon.info : 11[NET] received packet: from 213.233.85.41[11031] to 188.25.202.217[500] (580 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 11[ENC] parsed ID_PROT request 0 [ SA V V V V V V ]
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] received NAT-T (RFC 3947) vendor ID
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] received draft-ietf-ipsec-nat-t-ike-02 vendor ID
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] received draft-ietf-ipsec-nat-t-ike-02\n vendor ID
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] received draft-ietf-ipsec-nat-t-ike-00 vendor ID
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] received FRAGMENTATION vendor ID
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] received DPD vendor ID
Sat Jan 23 00:49:59 2016 daemon.info : 11[IKE] 213.233.85.41 is initiating a Main Mode IKE_SA
Sat Jan 23 00:49:59 2016 authpriv.info : 11[IKE] 213.233.85.41 is initiating a Main Mode IKE_SA
Sat Jan 23 00:49:59 2016 daemon.info : 11[ENC] generating ID_PROT response 0 [ SA V V V ]
Sat Jan 23 00:49:59 2016 daemon.info : 11[NET] sending packet: from 188.25.202.217[500] to 213.233.85.41[11031] (136 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 12[NET] received packet: from 213.233.85.41[11031] to 188.25.202.217[500] (228 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 12[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
Sat Jan 23 00:49:59 2016 daemon.info : 12[IKE] remote host is behind NAT
Sat Jan 23 00:49:59 2016 daemon.info : 12[ENC] generating ID_PROT response 0 [ KE No NAT-D NAT-D ]
Sat Jan 23 00:49:59 2016 daemon.info : 12[NET] sending packet: from 188.25.202.217[500] to 213.233.85.41[11031] (244 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 13[NET] received packet: from 213.233.85.41[11034] to 188.25.202.217[4500] (92 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 13[ENC] parsed ID_PROT request 0 [ ID HASH ]
Sat Jan 23 00:49:59 2016 daemon.info : 13[CFG] looking for pre-shared key peer configs matching 188.25.202.217...213.233.85.41[10.57.118.4]
Sat Jan 23 00:49:59 2016 daemon.info : 13[CFG] selected peer config "psk-l2tp"
Sat Jan 23 00:49:59 2016 daemon.info : 13[IKE] IKE_SA psk-l2tp[1] established between 188.25.202.217[188.25.202.217]...213.233.85.41[10.57.118.4]
Sat Jan 23 00:49:59 2016 authpriv.info : 13[IKE] IKE_SA psk-l2tp[1] established between 188.25.202.217[188.25.202.217]...213.233.85.41[10.57.118.4]
Sat Jan 23 00:49:59 2016 daemon.info : 13[ENC] generating ID_PROT response 0 [ ID HASH ]
Sat Jan 23 00:49:59 2016 daemon.info : 13[NET] sending packet: from 188.25.202.217[4500] to 213.233.85.41[11034] (76 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 15[NET] received packet: from 213.233.85.41[11034] to 188.25.202.217[4500] (108 bytes)
Sat Jan 23 00:49:59 2016 daemon.info : 15[ENC] parsed INFORMATIONAL_V1 request 2490515056 [ HASH N(INITIAL_CONTACT) ]
Sat Jan 23 00:50:00 2016 daemon.info : 16[NET] received packet: from 213.233.85.41[11034] to 188.25.202.217[4500] (460 bytes)
Sat Jan 23 00:50:00 2016 daemon.info : 16[ENC] parsed QUICK_MODE request 4000273885 [ HASH SA No ID ID ]
Sat Jan 23 00:50:00 2016 daemon.info : 16[IKE] expected IPComp proposal but peer did not send one, IPComp disabled
Sat Jan 23 00:50:00 2016 daemon.info : 16[IKE] received 28800s lifetime, configured 0s
Sat Jan 23 00:50:00 2016 daemon.info : 16[ENC] generating QUICK_MODE response 4000273885 [ HASH SA No ID ID NAT-OA NAT-OA ]
Sat Jan 23 00:50:00 2016 daemon.info : 16[NET] sending packet: from 188.25.202.217[4500] to 213.233.85.41[11034] (204 bytes)
Sat Jan 23 00:50:00 2016 daemon.info : 06[NET] received packet: from 213.233.85.41[11034] to 188.25.202.217[4500] (76 bytes)
Sat Jan 23 00:50:00 2016 daemon.info : 06[ENC] parsed QUICK_MODE request 4000273885 [ HASH ]
Sat Jan 23 00:50:00 2016 daemon.info : 06[KNL] received netlink error: No such file or directory (2)
Sat Jan 23 00:50:00 2016 daemon.info : 06[KNL] unable to add SAD entry with SPI ccc53882
Sat Jan 23 00:50:00 2016 daemon.info : 06[KNL] received netlink error: No such file or directory (2)
Sat Jan 23 00:50:00 2016 daemon.info : 06[KNL] unable to add SAD entry with SPI 03238277
Sat Jan 23 00:50:00 2016 daemon.info : 06[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Sat Jan 23 00:50:00 2016 daemon.info : 06[IKE] sending DELETE for ESP CHILD_SA with SPI 03238277
Sat Jan 23 00:50:00 2016 daemon.info : 06[ENC] generating INFORMATIONAL_V1 request 4148490349 [ HASH D ]
Sat Jan 23 00:50:00 2016 daemon.info : 06[NET] sending packet: from 188.25.202.217[4500] to 213.233.85.41[11034] (76 bytes)

stintel, 2016-02-29:
What OpenWrt version where you running? I had a similar problem, but this has been fixed in trunk: https://github.com/openwrt-mirror/openwrt/commit/5d40955aea8690ffcbdf57a7d7bfe61ef8bc11df

stintel, 2016-10-31:
@pynu please ignore my previous comment. The info I requested was already in the bug report. As the version you are running is newer than the one I mentioned, it's going to be a different problem.

The problem is that ``received netlink error: No such file or directory (2)`` is really vague, and I found it quite hard to debug. I eventually figured it out by adding many printk's in the kernel code used by strongswan.

Can you test if you still have this problem with strongswan 5.5.1? And if yes, please show me your ipsec.conf so I can try to reproduce this.

avbohemen, 2016-11-06:
I am having this exact problem in LEDE r2109 (kernel 4.4.30) using strongswan 5.5.1. My previous build with strongswan 5.5.0 (about 1 month old LEDE, slightly older 4.4 kernel)  worked fine.

I tried using ip-full instead of ip-tiny in LEDE, but no change.

My ipsec.conf:
```
# ipsec.conf - strongSwan IPsec configuration file
# basic configuration
config setup
        # strictcrlpolicy=yes
        # uniqueids = no

# Add connections here.
conn %default
        auto=add
        dpdaction=clear
        dpddelay=300s
        rekey=no
        left=%defaultroute
        leftid=router.fqdn.net
        leftsubnet=0.0.0.0/0,::/0
        leftfirewall=yes
        lefthostaccess=yes
        rightsourceip=%dhcp
        rightdns=192.168.196.100
        ike=aes256-sha1-modp1024,aes256-sha512-modp4096,aes128-sha1-modp1024,3des-sha1-modp1024!
        esp=aes256-sha256,aes256-sha1,3des-sha1!

conn ikev2-windows
        keyexchange=ikev2
        leftauth=pubkey
        leftcert=router.cer
        right=%any
        rightauth=eap-mschapv2
        rightsendcert=never

conn ikev2-iphone
        keyexchange=ikev2
        leftauth=psk
        right=%any
        rightid=iphone
        rightauth=psk

```

sbach, 2016-11-13:
I also had the same issue with kernel 4.4.27 and strongswan 5.5.1.

The issue can be fixed by applying this [1] patch from the Linux kernel mainline (see commit message for related discussion on the mailing list). This should generate the 'echainiv.ko' module that must be loaded before ipsec starts.

[1] https://github.com/torvalds/linux/commit/32b6170ca59ccf07d0e394561e54b2cd9726038c

avbohemen, 2016-11-14:
After reading the mailing list, I included kmod-crypto-echainiv in my LEDE image. This also solved the problem for me.

stintel, 2016-11-15:
@pynu does installing kmod-crypto-echainiv solve the problem for you? I am hesitant about adding a dependency for this, as it's only needed for CBC. I don't use CBC and my IPsec connections work just fine without the echainiv module.

In fact, kmod-ipsec depends on several modules that are not really required. I'd rather reduce the dependencies so people can do IPsec without adding unrequired bloat.

cbabs, 2016-11-17:
I'm on 5.5.1 on a November 14th 2016 trunk build of desg driv and have run into this.  I had to add kmod-crypto-echainiv after build because for some reason it did not install it even when I added the package explicitly.  I think this needs to be a dependency of strongswan as most use CBC.  This has just blown hours of my day for something silly.

stintel, 2016-11-17:
Those of you running LEDE, please test https://git.lede-project.org/?p=lede/stintel/staging.git;a=commit;h=d50c4aedf6e829ed16f4cdedae474ecfab67a4b8

fid0did0, 2016-11-26:
I'm experimenting the same problem.
Using the same "workaround" (installing kmod-crypto-echainiv) the problem is solved.

here my configuration:
root@lede:/etc# cat openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='HEAD'
DISTRIB_REVISION='r0+2321'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='ramips/rt305x'
DISTRIB_DESCRIPTION='LEDE Reboot r0+2321'
DISTRIB_TAINTS='no-all'

stintel, 2016-11-29:
Added a dependency on kmod-crypto-echainiv to kmod-ipsec in LEDE. Closing this issue, as it should not be fixed in the package feed.

closed 2016-11-29 by stintel