openwrt/packages #2300: ddns-scripts error Neither 'Wget' nor 'cURL' installed or executable

pynu, 2016-01-22:
OpenWrt Designated Driver r48455 / LuCI Master (git-16.020.59380-63d70da)
Kernel:4.4.0
Platform:X86_64, X86

Sat Jan 23 00:59:27 2016 user.notice ddns-scripts[10218]: noip: PID '10218' started at 2016-01-23 00:59
Sat Jan 23 00:59:27 2016 user.err ddns-scripts[10218]: noip: Neither 'Wget' nor 'cURL' installed or executable - TERMINATE
Sat Jan 23 00:59:27 2016 user.warn ddns-scripts[10218]: noip: PID '10218' exit WITH ERROR '1' at 2016-01-23 00:59

nbd168, 2016-01-22:
It needs a change to use uclient-fetch, like the one that I made for 6in4 here:
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=80cdacc0b9ae6c063499cc12cc62b7a083f62974

chris5560, 2016-01-26:
@pynu @nbd168
Fixed in ddns-scripts 2.6.0-1 merged yesterday.
Please close. I don't have permission to do.
Thanks Christian

closed 2016-01-26 by mhei

karlp, 2016-01-26:
wasn't it meant to be finding the symlink to wget? I mean, sure, this works, but the point was that wget was still meant to work right?