openwrt/packages #2308: ddns-scripts: incorrectly fails to verify valid DNS server

zxdavb, 2016-01-24:
Hi @chris5560,

I have a ddns configuration that works perfectly, unless I add (this is the authorative server):
```uci set ddns.namecheap.dns_server='dns1.registrar-servers.com'```

In which case, I get:
```
root@gl-router:~# cat /var/log/ddns/namecheap.log
172911       : ************ ************** ************** **************
172912  note : PID '5252' started at 2016-01-24 17:29
172912       : ddns version  : 2.5.0-1
172912       : uci configuration:
ddns.namecheap.dns_server='dns1.registrar-servers.com'
ddns.namecheap.domain='xxxxxx.me'
ddns.namecheap.enabled='1'
ddns.namecheap.interface='wan'
ddns.namecheap.ip_source='web'
ddns.namecheap.ip_url='http://myip.dnsomatic.com/'
ddns.namecheap.password='xxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
ddns.namecheap.service_name='namecheap.com'
ddns.namecheap.use_logfile='1'
ddns.namecheap.use_syslog='0'
ddns.namecheap.username='xxxxxxx'
ddns.namecheap=service
172912       : verbose mode  : 0 - run normal, NO console output
172913       : check interval: 600 seconds
172913       : force interval: 259200 seconds
172913       : retry interval: 60 seconds
172913       : retry counter : 0 times
172913       : No old process
172913       : last update: never
172913       : Verify DNS server 'dns1.registrar-servers.com'
172913       : #> /usr/bin/host -t ANY dns1.registrar-servers.com >/var/run/ddns/namecheap.dat 2>/var/run/ddns/namecheap.err
172913 ERROR : DNS Resolver Error - BIND host Error '1'
172913       :
172913  WARN : Verify DNS server 'dns1.registrar-servers.com' failed - retry 1/0 in 60 seconds
```

I note if I do the following:
```
root@gl-router:~# /usr/bin/host -t ANY dns1.registrar-servers.com
;; Truncated, retrying in TCP mode.
Host dns1.registrar-servers.com not found: 5(REFUSED)
```

But I also get:
```
root@gl-router:~# /usr/bin/host -t A dns1.registrar-servers.com
dns1.registrar-servers.com has address 216.87.155.33
```

chris5560, 2016-01-24:
Hi @zxdavb

i checked your

> |root@gl-router:~# /usr/bin/host -t ANY dns1.registrar-servers.com ;; Truncated, retrying in TCP mode Host dns1registrar-serverscom not found: 5(REFUSED)

on my wndr3800 running 15.05 (r48168) and on my virtualbox x86 running trunk (r48470) without any problems.

Sorry can't help, possibly a Bind host problem in your build !?

zxdavb, 2016-01-24:
Hmmm... I'm on 48396, but I have my own buildroot - it's not fancy, mind.

Who knows? I'll close the issue pending further investigation.  Any tips where to look?

closed 2016-01-24 by zxdavb

fbradyirl, 2017-04-27:
Hi zxdavb,

I recently came across a post from you on the openwrt [forum](https://forum.openwrt.org/viewtopic.php?id=50908&p=5): 
> my router is behind a cable modem, the modem has the public address, the router is a NAT client of the modem (192.168.1.1/24), and the router runs dnsmasq for the rest of the network (172.27.0.1/24).  So I thinks it's correct to use dnsomatic (i.e. the web option) to determine the address?
> 
> What I want is having the FQDN, db-router.no-ip.org, return the router's LAN address when my PCs/smartphones/etc are connected at home, and the public address otherwise (i.e. when connected via the Internet and therefore accessing my home network via the cable modem).
> 
> After reading your input (thanks!), I think I can sort most all of it at my end using **option dns_server**
> 

I am attempting to do the same thing and I am wondering how you got this set up and working on openwrt? 

I have been using Openwrt for a long time and am experienced with it but would love to find out how you set this up?

Thanks
Finbarr

chris5560, 2017-04-27:
Hi Finbarr,

to resolve db-router.no-ip.org from the lan with your local address you need to configure dnsmasq to give you the correct answer. i.e. using /etc/hosts

dnsomatic handle the global DNS servers if your wan ip changes.
dnsmasq handle your local DNS (with static IPs), so you need to configure locally.

chris5560, 2017-04-27:
Also have a look here https://forum.lede-project.org/t/dnsmasq-and-ddns-scripts/2563