openwrt/packages #2315: latest changes to gettext from mainline = smstools3 compile error

build000, 2016-01-26:
```
/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/toolchain-mipsel_1004kc+dsp_gcc-5.2.0_musl-1.1.12/lib/gcc/mipsel-openwrt-linux-musl/5.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: copy reloc against protected `__stack_chk_guard' is dangerous
charset.o: In function `iconv_convert.part.0':
charset.c:(.text+0x40): undefined reference to `libiconv'
charset.c:(.text+0xb0): undefined reference to `libiconv'
charset.o: In function `iconv_init':
charset.c:(.text+0xe52): undefined reference to `libiconv_open'
charset.c:(.text+0xe66): undefined reference to `libiconv_open'
collect2: error: ld returned 1 exit status
Makefile:51: recipe for target 'smsd' failed
make[5]: *** [smsd] Error 1
make[5]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/smstools3/src'
Makefile:10: recipe for target 'compile' failed
make[4]: *** [compile] Error 2
make[4]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/smstools3'
Makefile:68: recipe for target '/home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/smstools3/.built' failed
make[3]: *** [/home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/smstools3/.built] Error 2
make[3]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk/feeds/packages/utils/smstools3'
package/Makefile:191: recipe for target 'package/feeds/packages/smstools3/compile' failed
make[2]: *** [package/feeds/packages/smstools3/compile] Error 2
make[2]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk'
package/Makefile:188: recipe for target '/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.12/stamp/.package_compile' failed
make[1]: *** [/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.12/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk'
/home/slawek/build_openwrt/openwrt-witi-trunk/include/toplevel.mk:191: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
[13:23:33] slawek@slawek-dom:~/build_openwrt/openwrt-witi-trunk$ 
```

yousong, 2016-08-13:
ping maintainer of smstools3 @erdoukki ;)

build000, 2016-08-18:
latest changes - fixing problem - again build witch LEDE ... sorry - no more using openwrt, only LEDE ;)

closed 2016-08-18 by build000