openwrt/packages #2331: ddns-scripts updates every time for namecheap subdomains

kpu, 2016-01-31:
Namecheap can be used to update the IP of a subdomain, not just @ for a domain.  Let's say I want to update foo.example.com but not example.com.  The correct URL is

http://dynamicdns.park-your-domain.com/update?host=foo&domain=example.com&password=0123456789abcdef&ip=[IP]

To configure this with the OpenWRT URL format:

"namecheap.com" "http://dynamicdns.park-your-domain.com/update?host=[USERNAME]&domain=[DOMAIN]&password=[PASSWORD]&ip=[IP]"

I need to do 
username=foo
domain=example.com
password=somelongstring
That will update correctly.  However, the nslookup command polls example.com not foo.example.com, so the IPs never match.  And so the script does an update every tic.  

I've done a workaround by using a custom client and ignoring the domain field.

chris5560, 2016-01-31:
@kpu
What ddns-scripts/OpenWrt version are you working on ?
The current version of ddns-scripts on trunk has a separate entry for "lookup_host" which is used for nslookup. So USERNAME and DOMAIN and PASSWORD are simply url parameters.

kpu, 2016-01-31:
ddns-scripts - 2.4.3-1
luci-app-ddns - 2.2.4-1
OpenWrt Chaos Calmer 15.05
Looks like I just need to pull trunk then.

closed 2016-01-31 by kpu