openwrt/packages #2342: Prosody 0.9.9 cannot write /dev/urandom

dzavy, 2016-02-03:
This is a known issue (https://prosody.im/issues/issue/585), should be fixed in 0.9.10.

root@00E0C561182D:~# /etc/init.d/prosody start
lua: /usr/lib/prosody/util/uuid.lua:27: Unable to obtain a secure random number generator, please see https://prosody.im/doc/random (/dev/urandom: Permission denied)
stack traceback:
        [C]: in function 'error'
        /usr/lib/prosody/util/uuid.lua:27: in function 'uuid_gen'
        /usr/lib/prosody/core/hostmanager.lua:77: in function 'activate'
        /usr/lib/prosody/core/hostmanager.lua:44: in function '?'
        /usr/lib/prosody/util/events.lua:67: in function 'fire_event'
        /usr/lib/prosody/../../bin/prosody:337: in function 'prepare_to_start'
        /usr/lib/prosody/../../bin/prosody:403: in main chunk
        [C]: ?

MikePetullo, 2016-09-15:
This should be fixed, as prosody 0.9.10 no longer opens /dev/urandom for writing, only reading.

diizzyy, 2017-06-09:
@dzavy 
Still an issue?

closed 2017-08-16 by heil