openwrt/packages #2347: mwan3 udp

kubco2, 2016-02-05:
Hello,
Im trying udp openvpn server with 2ifaces in mwan3.
iface1  - metric 1
iface2  - metric 2
But if I connect over iface2, answer goes over iface1.

With tcp openvpn server it works out of box.

Is there missing some modules for udp tracking?

kubco2, 2016-02-05:
Probably it is not problem of mwan3 ..

mstorchak, 2016-02-05:
Do you have active-backup or load balancing configuration in mwan3? I used openvpn-udp with mwan3-loadbalancing and it worked perfectly.

kubco2, 2016-02-05:
only backup link, no loadbalancing.

mstorchak, 2016-02-05:
Perhaps you should try load balancing. If you still want the end result to look like active/backup, set weights to something like 1:10000 (I don't really know what's the upper limit for weight).
The reason is that in active/backup case ALL traffic goes through one interface, because there is a single route to pick from - the one with the lowest metric. Although this doesn't explain why tcp works for you, maybe NAT handles udp and tcp differently enough to make that happen.

kubco2, 2016-02-06:
I cant change it to loadbalancing because there is LTE modem with additional charges, so 0.00001% is a always risk.

Problem is that MWAN3 doesnt pick existing "reverse" connection from conntrack, rather create new one and if it is new then it goes over primary route.

It doesnt have to be MWAN3 problem, but conntrack problem or if it cannot be handled by conntrack then it is property of UDP stateless protocol.

champtar, 2016-02-06:
Hi @kubco2,
what you can do is make openvpn listen on port 10001 on wan1 and 10002 on wan2 and you add rule in mwan3 to say outgoing local udp traffic with source port 10001 go through wan1 and outgoing local udp traffic with source port 10002 go through wan2.

yousong, 2016-08-13:
More like a configuration issue.  Any updates on this one?

jsalatiel, 2017-04-12:
I dont believe it is a configuration issue. I face the same problem. TCP works , udp fails if mwan3 is enabled.
One workaround is use 'option local IP_FROM_SOME_INTERFACE' and it will work for that interface.

feckert, 2018-11-30:
@kubco2 could we close this issue?
It´s very old and there were a lot of changes in openwrt master.

kubco2, 2018-12-03:
I would like to try it , but later this month.

feckert, 2019-03-28:
No reaction or feeback for 3 month from issue creator so I will close this

closed 2019-03-28 by feckert