openwrt/packages #2348: ddns: Namecheap can easily work for non-matching FQDN

dsjstc, 2016-02-05:
For Namecheap DDNS, the functions script only works if the IP for the dynamic domain matches the IP for the TLD.  This is not a common use case.

It was trivial to modify dynamic_dns_functions.sh so that the DNS lookups are to the targetted dynamic domain.  All that is necessary is to change every occurrence of $domain to $username.$domain, except for the first one ( s#\[DOMAIN\]#$domain#g ).

I'm afraid I don't know how to make a ddns-scripts_namecheap package to distribute this change.

chris5560, 2016-02-06:
I never touch namecheap support inside ddns-scripts. 
The only thing I found out is that the discription at 

https://www.namecheap.com/support/knowledgebase/article.aspx/29/11/how-do-i-use-the-browser-to-dynamically-update-hosts-ip

say the update url should look like

https://dynamicdns.park-your-domain.com/update?host=[DOMAIN]&domain=[USERNAME]&password=[PASSWORD]&ip=[IP]

inside /usr/lib/ddns/services the url is since a long time

http://dynamicdns.park-your-domain.com/update?host=[USERNAME]&domain=[DOMAIN]&password=[PASSWORD]&ip=[IP]

what is the right one ? host=[DOMAIN] or host=[USERNAME]

Beginning ddns-scripts 2.5.x [DOMAIN] is no longer used to validate an IP update via nslookup.
Also additional parameters are supported which can be used inside update url.

Please help me to fix this because I don't have an account at namecheap.
Give me an example of your needs with a complete update url.

Christian

dsjstc, 2016-02-06:
The script has no HOSTNAME variable, so USERNAME is used to hold the 
hostname.  The current url correctly requests an update for dynamic host 
USERNAME.DOMAIN.

The problem is that somewhere in the script incorrectly assumes that 
nslookup(DOMAIN) will return the ip address of the dynamic host. 
Changing all the lookups from nslookup(DOMAIN) to 
nslookup(USERNAME.DOMAIN) is what corrected my problem.

Probably the general solution is to create a new variable FQDN which is 
allowed to be different from DOMAIN.  Possibly we also need a new 
HOSTNAME input.  Let me know if that doesn't make sense and I'll try to 
make a patch next week.

chris5560, 2016-02-06:
Which ddns-scripts version are you talking about ?

Since Version 2.5.x (available on trunk) there are new options inside /etc/config/ddns available
- option lookup_host => the FQDN used by nslookup to validate IP, so option domain [DOMAIN] is no longer used to validate IP.
- option param_enc => to be used as additional (urlencoded) parameter [PARAMENC] inside update url.
- option param_opt => same as above but will not be urlencoded.

So this should fulfil your requirements. To use a different "lookup_host" then defined in "domain", "username" or any parameter inside update url.

dsjstc, 2016-02-06:
Uh-oh!  New to openwrt, just assumed that the opkg version was most 
current.  Will review when I return home.

dsjstc, 2016-02-12:
I was indeed working with ddns-scripts v2.4 from the 15.05 Chaos release.  It sounds like the new options will solve the described problem, so I'll close the issue.

closed 2016-02-12 by dsjstc