openwrt/packages #2349: luci-app-mwan3 spamming syslog

psyborg55, 2016-02-05:
the number of mwan3 configured interfaces equals to number of error messages each 5 seconds. it happens on main page and mwan overview page so I guess it has something to do with MWAN Interface Live Status

more info: https://dev.openwrt.org/ticket/19419

hnyman, 2016-02-06:
cc to maintainers of luci-app-mwan3 and mwan3
@arfett @Adze1502

arfett, 2016-02-27:
I'll look at this. Eventually.

psyborg55, 2017-02-05:
replacing all "uci -p" with "uci -q -p" in mwan3.lua silences output of these messages to syslog. is there any better solution to the problem that would affect script efficiency?

arfett, 2017-02-06:
@psyborg55, If you can allow me SSH to your router I can take a look. E-mail me ch i p d ank ly at gm ail.co m

psyborg55, 2017-02-12:
found a solution meanwhile and opened PR https://github.com/openwrt/packages/pull/3986

closed 2017-02-12 by psyborg55