openwrt/packages #2350: airmon-ng outputs junk to console

psyborg55, 2016-02-06:
The issue has been existed since BB, it prints out some usage info and worse it treats wifi interfaces properties as interfaces. 2 seconds job takes 10 or more seconds.

```
root@OpenWrt:~# airmon-ng start wlan2
ps: unrecognized option: A
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.24.1 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output

Interface	Chipset		Driver

wlan1-1		Unknown 	rt2800_wmac - [phy1]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11bgn		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=20		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan0-3		Unknown 	mt76x2e - [phy0]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11an		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=17		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan2		Atheros AR9271	ath9k - [phy2]
				(monitor mode enabled on mon0)
wlan0-2		Unknown 	mt76x2e - [phy0]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11an		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=17		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan1-4		Unknown 	rt2800_wmac - [phy1]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11bgn		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=20		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan1		Unknown 	rt2800_wmac - [phy1]
wlan0-1		Unknown 	mt76x2e - [phy0]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11an		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=17		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan1-3		Unknown 	rt2800_wmac - [phy1]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11bgn		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=20		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan0		Unknown 	mt76x2e - [phy0]
wlan1-2		Unknown 	rt2800_wmac - [phy1]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11bgn		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=20		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
wlan0-4		Unknown 	mt76x2e - [phy0]
IEEE		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
802.11an		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Mode:Master		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
Tx-Power=17		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
dBm		Unknown		Unknown (MONITOR MODE NOT SUPPORTED)
```

psyborg55, 2016-02-08:
aircrack part of the issue is resolved by replacing

for iface in `iwconfig 2>/dev/null | egrep '(IEEE|ESSID|802\.11|WLAN)' | sed 's/^\([a-zA-Z0-9_.]*\) .*/\1/' | grep -v wifi`

with

for iface in `iwconfig 2>/dev/null | egrep '(IEEE|ESSID|802\.11|WLAN)' | sed 's/^\([a-zA-Z0-9_-]*\) .*/\1/' | grep -v wifi`

but the busybox still complains about ps: unrecognized option: a

yousong, 2016-02-09:
Hi, it is expected that busybox ps does not work with airmon-ng.  You can add a monitor mode interface by following instructions at https://github.com/openwrt/packages/pull/1693#issuecomment-133818911.

We are about to split airmon-ng out from aircrack-ng as a separate pacakge.

psyborg55, 2016-02-09:
Hi. Thanks for replying. I've already made some modifications to airmon-ng and sent the patch to the mailing list. With modified script the output is as expected, without any garbage.

Further, to work in monitor mode with newer airmon-zc or iw, you must bring the interface down and then you are limited to monitor mode only because of channel -1 issue.

The greatest disadvantage of this is I can't no longer monitor active wifi link using airodump-ng.

I have a solution for channel -1 too, if you are interested I'll post a patch (it fixes the issue at mac80211 level) so you can keep running airmon-ng/airodump-ng with sta interface and as many mon interfaces as required just like before the busybox update broke this!

I'd like to keep it this way as it's been undoubtedly the simplest one.

yousong, 2016-02-10:
> Hi. Thanks for replying. I've already made some modifications to airmon-ng and sent the patch to the mailing list. With modified script the output is as expected, without any garbage.

I have wondered several times in the past whether aircrack-ng has a mailing list for patches, and thought http://github.com/aircrack-ng/aircrack-ng should be one of the official repo.

> Further, to work in monitor mode with newer airmon-zc or iw, you must bring the interface down and then you are limited to monitor mode only because of channel -1 issue.
>
> The greatest disadvantage [...]

The patch can be interesting.   I think you can post it here just for archive purposes.

> 
> I'd like to keep it this way as it's been undoubtedly the simplest one.
> 

Yeah, as long as it works :)

psyborg55, 2016-02-10:
First patch to fix airmon-ng

[airmon-ng.txt](https://github.com/openwrt/packages/files/124705/airmon-ng.txt)

It is also posted at https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg37560.html but it didn't make it yet to patchwork site...no idea why...

And second one for channel -1 (it might not apply correctly to latest trunk because it was created quite some time ago)

[backports.txt](https://github.com/openwrt/packages/files/124707/backports.txt)

yousong, 2016-02-10:
I guess the reason of the patch not getting picked up by patchwork is that the patch was not of`text/plain` type (plus it's line wrapped and possibly tabs getting replaced with spaces).  git-send-email should do the job.

But patches for packages feed should go here anyway, and better yet patches for fixing packages themselves should also get a chance to receive upstream's review/comments first :)

psyborg55, 2016-02-10:
I've followed wiki instructions when sending patches, so there should be no issues with patch format. Is git-send-email a must?

yousong, 2016-02-11:
No, `git-send-email` is not a must, to me it's just the easier tool to send patches around.

I check the mail `[PATCH] packages: fix airmon-ng output in multi bssid mode` with `Show original` of web gmail and list archive at https://lists.openwrt.org/pipermail/openwrt-devel/2016-February/039512.html, both indicated that the patch was mangled.

psyborg55, 2016-02-11:
You are right. Damn gmail, in my Sent Mail patches show up as they were created, but as soon as I click Show original they're whitespace damaged even though the source code of the email shows "white-space:pre".

Additionally, the airmon-ng patch had no whitespace at all because the script source code was already whitespace broken, is there some special way to deal with such patches?

yousong, 2016-02-12:
Try following the original code style, it's fine as long as the patch applies.  Anyway, you may want to try aircrack-ng upstream first as airmon-ng has changed quite a lot since 1.2-rc1.

psyborg55, 2016-02-12:
tried upstream already and it's even more broken

yousong, 2016-02-12:
> tried upstream already and it's even more broken

Hah, it's -rcX anyway, not quite unexpected...

psyborg55, 2016-02-18:
Hi

I've resent patches using mail client, still they aren't picked up by ozlabs. It would be nice if those were included in 15.05.1 while we wait for its release.

yousong, 2016-02-23:
Hi, @psyborg55 

Patches for aircrack-ng should go here via pull request.  See https://wiki.openwrt.org/doc/devel/packages#how_to_submit_patches_to_openwrt for details

As for those patches you sent to openwrt-devel not being picked up by patchwork, I am not sure if patchwork can handle base64-encoded mail, but the patch was still line-wrapped if you inspect the content by decoding with `base64 -d`.  But this is another problem when you need to send patches to the list.

yousong, 2016-08-13:
Any news on this one?

psyborg55, 2016-09-13:
yeah. where is that worksforme problem resolution?

latest trunk compiled, busybox spam no longer appears, but multissid airmon issue is still there unless replaced . with -

yousong, 2016-09-13:
Please consider opening a pr about that fix, preferably to upstream aircrack-ng first.  The busybox `ps` problem was probably fixed by depending on `procps-ng` instead of `procps` (just for archive purposes, the pr is at https://github.com/openwrt/packages/pull/2873)

yousong, 2016-12-17:
Close this for now as airmon-ng now explicitly depends on procps-ng and airmon-ng is not absolutely needed in current OpenWrt/LEDE environment anyway

closed 2016-12-17 by yousong

psyborg55, 2017-01-22:
this won't work upstream. these are two different bugs: mainline version needs to have their interfaces named wlan0.mon or something like that, with multibssid case we have wlan0-1

and please stop complicating with this stuff, to run airmon-zc, it will depend on lsusb which will add another several KBs to final image, if in the end it will work correctly at all...

please stick to the tested and known to work solutions