openwrt/packages #2355: lighttpd: module loading missing symbols

githoc, 2016-02-07:
The stable Chaos Calmer 15.05 sunxi image/packages includes a lighttpd which fails to load the mod_redirect.so module, and possibly others.

/usr/sbin/lighttpd: symbol 'buffer_copy_string_buffer': can't resolve symbol
2016-02-06 14:31:31: (plugin.c.169) dlopen() failed for: /usr/lib/lighttpd/mod_redirect.so  
2016-02-06 14:31:31: (server.c.679) loading plugins finally failed 

The stable image includes lighttpd version : 
root@hoc:~# lighttpd -v
lighttpd/1.4.36 (ssl) - a light and fast webserver
Build-Date: Sep  8 2015 14:43:01

An older Chaos Calmer snapshot (15.05-rc1, r45695) doesn't have this problem. This snapshot comes with lighttpd version : 
root@hoc:~# lighttpd -v
lighttpd/1.4.35 (ssl) - a light and fast webserver
Build-Date: May 14 2015 01:00:43

yousong, 2016-08-13:
ping maintainer of lighthttpd @MikePetullo ;)

gstrauss, 2017-01-05:
@MikePetullo I believe this is fixed in latest openwrt/packages.  Not sure about in Chaos Calmer 15.05.1

diizzyy, 2017-06-09:
@yousong 
Can you please close this?

closed 2017-06-09 by yousong