openwrt/packages #2364: Nginx - proxy_pass

adrienb4, 2016-02-10:
Hello

I try to understand why when I use Nginx (1.9.9) with a proxy_pass, the response headers are in the body and not in the resquest. With the same conf/version in Debian 8, I have no problem.

**Web page :**
```
Server: nginx/1.6.2
Date: Wed, 10 Feb 2016 08:19:37 GMT
Content-Type: text/plain
Content-Length: 6
Last-Modified: Fri, 05 Feb 2016 14:38:29 GMT
Connection: close
ETag: "56b4b3e5-6"
Accept-Ranges: bytes

test
```
**Nginx log :**
```
2016/02/10 08:19:31 [error] 1467#0: *2 upstream sent no valid HTTP/1.0 header while reading response header from upstream, client: 192.168.1.146, server: localhost, request: "GET /favicon.ico HTTP/1.1", upstream: "http://192.168.6.40:80/favicon.ico", host: "192.168.1.1:8080", referrer: "http://192.168.1.1:8080/test.txt"
```

Any idea to solve this ?

Thank

Adrien

heil, 2016-02-10:
please give a link to your obfuscated config

yousong, 2016-02-10:
NGINX failed parsing status line from upstream server and defaulted to pass the content from upstream as HTTP/0.9 payload.  You may want to sniff with traffic between NGINX and upstream server with `tcpdump` to firstly make sure the status line is alright.

adrienb4, 2016-02-10:
Thank for your reponses. I solved my problem to download the last sources and build nginx in 1.9.9.
I think the problem is that I used the sources (release 15.05) with Nginx 1.9.9

adrienb4, 2016-02-11:
I just try again. I confirm, with the release v15.05 and Nginx 1.9.9 (or 1.9.6) the proxy_pass doesn't work.

**Verry basic conf :**
```
server {
        listen   80;
        server_name 192.168.1.1;
        root /etc/nginx/crossdomain/;
        #resolver 208.67.222.222;
        resolver 8.8.8.8;
        access_log /var/log/nginx/access.log;
        error_log /var/log/nginx/error.log;

        location / {
                proxy_pass http://192.168.6.40/;
        }
}
```

adrienb4, 2016-02-11:
I compiled Nginx 1.4.7 in Openwrt 15.05 sources and I have an another problem.

```
192.168.1.146 - - [11/Feb/2016:10:08:11 +0000] "\xC8\xE7\x16\x00\xD4\x15\xC8\xB6t.txt HTTP/1.1" 400 172 "-" "-"
192.168.1.146 - - [11/Feb/2016:10:09:11 +0000] "-" 408 0 "-" "-"
```

Why my URL is en hex ?? 

Thank

Adrien

adrienb4, 2016-02-11:
The HEX log was corrected by adding "Enable  HAProxy proxyprotocol" for Nginx in sources.
But... same problem as before. The headers are in the body.

adrienb4, 2016-02-11:
Attached tcpdump file. Rename in .pcap
[nginx.txt](https://github.com/openwrt/packages/files/126477/nginx.txt)

yousong, 2016-02-11:
I see no status line from `192.168.1.1.80` in the pcap file, so I guess it's more likely a upstream server problem.  You can confirm this by `proxy_pass` location `/` to a known working site, e.g. a local site with `python -mSimpleHTTPServer`.

adrienb4, 2016-02-11:
Yes the server (192.168.6.40) in proxy_pass work with no problem

**http://192.168.6.40/test.txt - (ok)**
```
prout
```

**http://192.168.1.1/test.txt - (not ok)**
```
Server: nginx/1.6.2
Date: Thu, 11 Feb 2016 14:02:17 GMT
Content-Type: text/plain
Content-Length: 6
Last-Modified: Fri, 05 Feb 2016 14:38:29 GMT
Connection: close
ETag: "56b4b3e5-6"
Accept-Ranges: bytes

prout
```

yousong, 2016-02-11:
Sorry, I misunderstood some roles of each ip address in the .pcap file.

You need to check the traffic between 192.168.1.1:80 and 192.168.6.40:80.  It is the response of 192.168.6.40:80 not recognised by 192.168.1.1:80.  But I really have no idea why it worked fine if accessed directly.  Does the problem persist if we `proxy_pass` to another site instead of `192.168.6.40:80`?

adrienb4, 2016-02-11:
Yes the problem presist with another website. I have no idea to solve this.
All is ok the the last sources of Openwrt and nginx.

yousong, 2016-02-11:
Hmm, I just tried NGINX/1.9.9 from master branch here with latest of 15.05 (http://git.openwrt.org/?p=15.05/openwrt.git;a=commit;h=c698aa66043a151ac76d19849be9ee24dfd78b72), `proxy_pass` works fine...

adrienb4, 2016-02-12:
You are lucky. I have tested with your commit and I have the same error. I use Asus AC56U router.

yousong, 2016-02-12:
I use qemu malta big endian machine.  You may want to try a clean build, i.e. `make dirclean`

adrienb4, 2016-02-12:
I just clone git from SVN. The build is in progress. Wait and see...

adrienb4, 2016-02-12:
Same problem. I test on Netgear R7000 now.

yousong, 2016-02-12:
Hmm, checking the traffic between nginx and upstream server should help make sure whether it's really a problem at NGINX side.

adrienb4, 2016-02-12:
I have no problem with nginx and the upstream. The upstream respond correctly. However I have no problem this the sources in master branch (github). The config is ths same.

adrienb4, 2016-02-12:
I have no idea to solve this. 
It seems to miss the field "status line" in the response upstream. But how add this.....

Any help please :)

yousong, 2016-02-13:
Well, now we know it's a problem at the upstream side.  But let's clarify a few points about the previous observations first

1. Why it worked fine if you access it directly?  What's the differences between requests sent by NGINX and the tool you use.
2. Why `the problem presist with another website`?  There are so many sites in the wild and we are this lucky to hit another one with the same problem?

adrienb4, 2016-02-16:
I dont think it's a problem at the upstream side. For each upstream I test the result is the same. From my computer, I use the same Nginx version and the same configuration and I have no problems.

Adrien

heil, 2017-08-16:
I will check now with 1.12.1 and lede to see if this config still has issues

```
server {
        listen   80;
        server_name 192.168.1.1;
        root /etc/nginx/crossdomain/;
        #resolver 208.67.222.222;
        resolver 8.8.8.8;
        access_log /var/log/nginx/access.log;
        error_log /var/log/nginx/error.log;

        location / {
                proxy_pass http://192.168.6.40/;
        }
}

```

closed 2018-06-12 by heil