openwrt/packages #2369: eudev builds error - not applying properly patch to mtd

build000, 2016-02-10:
```
(...)
make[3]: Entering directory '/home/slawek/build_openwrt/openwrt-witi-trunk/feeds/packages/utils/eudev'
. /home/slawek/build_openwrt/openwrt-witi-trunk/include/shell.sh; gzip -dc /home/slawek/build_openwrt/openwrt-witi-trunk/dl/eudev-3.1.5.tar.gz | tar -C /home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/eudev-3.1.5/.. -xf - 

Applying ./patches/0001-mtd_probe-uses-stdint_h.patch using plaintext: 
patching file src/mtd_probe/mtd_probe.h

Applying ./patches/0003-add_btn_trigger_happy_define.patch using plaintext: 
patching file src/udev/udev-builtin-input_id.c

Applying ./patches/0004-mtd_probe-uses-stdint_h.patch using plaintext: 
patching file src/mtd_probe/mtd_probe.h
Hunk #1 FAILED at 18.
1 out of 1 hunk FAILED -- saving rejects to file src/mtd_probe/mtd_probe.h.rej
Patch failed!  Please fix ./patches/0004-mtd_probe-uses-stdint_h.patch!
Makefile:140: recipe for target '/home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/eudev-3.1.5/.prepared_160f1fa2847e2b8a930e98e0abf4be7c' failed
make[3]: *** [/home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.12/eudev-3.1.5/.prepared_160f1fa2847e2b8a930e98e0abf4be7c] Error 1
make[3]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk/feeds/packages/utils/eudev'
package/Makefile:192: recipe for target 'package/feeds/packages/eudev/compile' failed
make[2]: *** [package/feeds/packages/eudev/compile] Error 2
make[2]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk'
package/Makefile:189: recipe for target '/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.12/stamp/.package_compile' failed
make[1]: *** [/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.12/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk'
/home/slawek/build_openwrt/openwrt-witi-trunk/include/toplevel.mk:192: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
[20:28:13] slawek@slawek-dom:~/build_openwrt/openwrt-witi-tr
```

dangowrt, 2016-02-10:
Should be fixed by https://github.com/openwrt/packages/commit/97d0a68ef85fc32e03a916dd20e9f6ece32dbdb2
Please retry and let me know if that solves it.

build000, 2016-02-10:
confirm:
https://github.com/openwrt/packages/commit/97d0a68ef85fc32e03a916dd20e9f6ece32dbdb2
- builds problem is fixed on this patch.
As for the proper operation of the (eudev) substitute udev - for the time being not to speak (I do not know when I'll have time to test it, in my case usbip).

dangowrt, 2016-02-10:
thanks for testing.
eudev is basically just a fork of systemd-udev-220 with all the systemd-related stuff removed. it should work just as well as udev-173 did before.
usbip is the only user other than libinput, so I've already updated the dependencies to unbreak build now that udev was removed from core.

closed 2016-02-10 by dangowrt