openwrt/packages #2373: glib2 / vala: fails to run with LANG=C

plntyk, 2016-02-11:
found via broken seafile-ccnet on multiple platforms (this package has issues and probably is to be deleted but the error should still be fixed):

http://buildbot.openwrt.org:8010/broken_packages/x86.64/seafile-ccnet/compile.txt

```
valac --pkg posix ccnetobj.vala -C -H ccnet-object.h
GLib: Cannot convert message: Could not open converter from 'UTF-8' to 'ANSI_X3.4-1968'

** (valac:23788): CRITICAL **: file valacompiler.c: line 2594: unexpected error: Could not open converter from 'ANSI_X3.4-1968' to 'UTF-8' (g_convert_error, 2)
```
The culprit seems to be the environment variable LANG=C.
manually: OpenWrt valac in staging_dir/.../host/ fails with LANG=C
manually: OpenWrt valac in staging_dir/.../host/ runs with LANG=de_DE.UTF-8
manually: run host valac sucessfully with any LANG option set (Archlinux with vala 0.30.1)

Either a vala error or more likely a glib2 error (g_convert_error).

kissg1988, 2016-04-30:
I'm currently in the process of packaging seafile server and have stumbled upon this issue. 

Seems like this is caused by the libiconv stub of openwrt, it does not support the charset 'ANSI_X3.4-1968' therefore the conversion fails.

Some of the possible solutions:
* build the host version of glib2 against libiconv-full instead of the libiconv stub
* patch the libiconv stub to support ANSI_X3.4-1968 and possibly other ASCII-compatible charsets
* patch glib2 to treat all ASCII aliases as 'US-ASCII' which is supported by the stub
* set a UTF-8 locale (eg. "en_US.UTF-8") somewhere in the package definition to avoid conversion (this is more like avoiding issue rather than fixing it, though)

I'll try to figure out which one of the above works and provide a fix soon.

kissg1988, 2016-05-02:
I've created a slight patch for the libiconv stub and sent it to openwrt-devel list. That should fix this issue.

kissg1988, 2017-10-04:
@plntyk 
This has been fixed over a year ago, would you please close the issue? Thanks!

cshoredaniel, 2018-07-30:
ping on closing this

closed 2018-08-17 by dibdot