openwrt/packages #2395: Failure to detect running WAN interface when driver reports "unkown" state

Slabity, 2016-02-19:
I'm running an Archer C7 with OpenWRT version DD r48648. When I try to run the script, it hangs and then reports the following error:

`adblock[....]` error: wan update interface/device not running (wan/pppoe-wan), rc: 145`

However, the interface is running and I have full access to the outer network.

Looking more carefully, it appears that `cat /sys/class/net/pppoe-wan/operstate`  reports "unkown", while the script expects it to report "up". I'm not sure why, but I've done some digging and I found this on the Debian mailing list:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=539583#37

If it's true, that means that some interfaces will always report "unknown" on a fully working interface.

I was able to solve this issue by changing `adblock-helper.sh` to also allow for interfaces to check for the "unknown" state. However, I'm not sure if this is the best solution.

hnyman, 2016-02-19:
cc to maintainer: @dibdot

hnyman, 2016-02-19:
@Slabity 
I renamed this issue for adblock, as that package seems to suffer from the underlying bug. The actual underlying bug might be located in netifd or pppoe modules, which are handled in the main Openwrt bug tracker, not here at the packages feed, (or the bug might even be in the kernel itself).

@dibdot should be able to fix adblock to operate with pppoe wan interfaces despite their wrong status info.

closed 2016-02-19 by Slabity

Slabity, 2016-02-19:
Marking as closed. Thanks for the fix @dibdot 

However, I'm a bit concerned with the idea of using a ping to test connectivity. It's very possible someone blocks ICMP (common security measure), and I don't think we would want to rely on a hardcoded IP address (even if it is 8.8.8.8).

dibdot, 2016-02-19:
@Slabity 
The hardcoded ip addresses will be removed with the next regular release. I have no better idea for a uplink test (if a local check is not possible/reliable), feel free to propose a better approach. Relying on an "unknown" device status is no solution ...

Slabity, 2016-02-19:
What about a DNS request? DNS is almost never completely blocked on the network. It's also likely to be required to resolve the names for the filters anyways.

But you're right. I don't see a clean way to do it.