openwrt/packages #2401: emailrelay crashes

rodajc, 2016-02-21:
As a server emailrelay receives mail from an smtp server. But after receiving one message emailrelay exits with a segmentation fault.
In fact, emailrelay seems to crash every time it needs to disconnect. Also in client mode and in POP mode.
In POP mode I see the message received in Thunderbird with the Fetch header only option enabled. With this option disabled Thunderbird starts downloading the message and continues to do so until you abort it manually. The Inbox grows and grows with just CRLF characters.
The reason is that emailrelay seems to be unable to disconnect and if the peer does, it crashes with a segmentation fault.
Tested with trunk revision 48679 and [this router](https://forum.openwrt.org/viewtopic.php?id=56382).

rodajc, 2016-05-09:
Closing was a bit premature. After testing the other modes these problems remain in revision 49263.
In POP mode the problems remains that I see the message received in Thunderbird with the Fetch header only option enabled. With this option disabled Thunderbird starts downloading the message and continues to do so until you abort it manually. The Inbox grows and grows with just CRLF characters.
In server mode the --filter switch doesn't seem to work.

yousong, 2016-08-11:
ping emailrelay's maintainer @fededim ;)

fededim, 2016-08-11:
Reply from maintaner @fededim: bytes=xx time=21600s TTL=53 :)

Emailrelay is already updated to latest version 1.9 (see https://sourceforge.net/projects/emailrelay/files/ )..can you please test it on a linux pc (use it as server) just to be sure it's an issue of openwrt toolchain and not of the software ? I have used it mainly as smtp relay server, not as pop3 server.

fededim, 2016-08-11:
I have seen this crash log after disconnection (I am wondering why netcat is able to connect to port 25 if the specified listening port is 10025 nevertheless) at https://forum.openwrt.org/viewtopic.php?id=62639, probably it's the same as yours...it seems it's a libc problem (so due to openwrt toolchain), but it is difficult to tell which is the call which makes the app crash. The last log line emailrelay outputs is "emailrelay: info: smtp connection closed: read error: disconnected: 127.0.0.1:34584" which can be found in file gsmtpserver.cpp around line 105

void GSmtp::ServerPeer::onDelete( const std::string & reason )
{
	G_LOG_S( "GSmtp::ServerPeer: smtp connection closed: " << reason << (reason.empty()?"":": ")
		<< peerAddress().second.displayString() ) ;

	m_server.eventSignal().emit( "done" , reason ) ;
}

Unluckily the function quits immediately and without a debbuger it's difficult to tell which code is executed after (probably it can be done through gdb remote debugging on router if the right packages are available, but I only know Windows programming, not linux embedded, if someone can help with this he is welcome).

Try to turn on --verbose or better the --debug log option in the command line used to launch emailrelay and post the log here. Debug logging in order to work needs to enabled at compile time by passing --enable-debug options to configure which currently is not (you can compile emailrelay by yourself by downloading openwrt sources, editing its Makefile included in the local directory packages/net/emailrelay/Makefile (e.g. https://github.com/openwrt/packages/blob/master/net/emailrelay/Makefile) by adding at line 83-84 line with 

CONFIGURE_ARGS += \
	--with-pam=no \
	--enable-debug=yes
)

I have made the changes directly from github website without testing (I can't do it now, it shuold compile however) and made a PR (https://github.com/openwrt/packages/pull/3045), wait till it will be merged, then wait one more day in order to openwrt buildbot to generate the new emailrelay package, reinstall it in your router and launch emailrelay with --debug log option.

If you are willing to wait or it does not compile you can try with --verbose log and post log here.

p.s. remember to always post the log when submitting an issue.

fededim, 2016-08-11:
Just a minor update, in hurry I wrongly reported the last code executed before the error, the correct one is in gsocketprotocol.cpp, in any case it's better to have a full debug log before digging into the issue so let's wait for the pr to be merged.

```
void GNet::SocketProtocolImp::rawReadEvent()
{
	char buffer[c_buffer_size] = { '\0' } ;
	const size_t buffer_size = G::Test::enabled("small-client-input-buffer") ? 3 : sizeof(buffer) ;
	const ssize_t rc = m_socket.read( buffer , buffer_size ) ;

	if( rc == 0 || ( rc == -1 && !m_socket.eWouldBlock() ) )
	{
		throw SocketProtocol::ReadError() ;    // this one
	}
	else if( rc != -1 )
	{
		G_ASSERT( static_cast<size_t>(rc) <= buffer_size ) ;
		m_sink.onData( buffer , static_cast<std::string::size_type>(rc) ) ;
	}
	else
	{
		; // -1 && eWouldBlock() -- no-op (esp. for windows)
	}
}
```

fededim, 2016-09-15:
Hi again it seems that PR can't be merged, rodajc do you know how to compile the email relay package with the changes inside my pr ? If not I'll provide you a new package with support for extended debug, let me know.

rodajc, 2016-09-16:
Hi Federico,

I know how to compile OpenWrt but I am not an expert, to say the least. 
If you can tell me where to copy the sources or the package I can give 
it a try.

Thank you very much.

Kind regards,

Suso

fededim, 2016-09-18:
I am updating the Makefile in order to allow the debugging logging as an option in make menuconfig (as default it will disabled), in this way it would be easy for you to recompile with debug support (sorry for delay but I am experiencing some compilation errors). I'll give you an update.

fededim, 2016-09-18:
New PR done https://github.com/openwrt/packages/pull/3185/files. If you do not want to wait, you can:

1) grab a copy of the new [Makefile](https://github.com/fededim/packages/blob/26b28fe48d3af369f669f741d2d8f868a7e77cc0/mail/emailrelay/Makefile) and replace your Makefile in openwrt/feeds/packages/net/emailrelay/Makefile.
2) Move the folder openwrt/feeds/packages/net/emailrelay to <openwrt folder>/feeds/packages/mail/emailrelay.

Pay attention that now you'll find emailrelay **not in the Network menu of make menuconfig, but in Mail** (don't ask why I put it the network category because I can't remember, probably I was drunk :-). If you select the package you'll have now the option Enable support for extended logging. Enable that, build and install the new package. When starting emailrelay pass --debug as commandline option and post crash log here.

rodajc, 2016-09-22:
Hi Federico,

Thank you for your effort and for your reply!

Following your instructions the emailrelay package doesn't show up in 
menuconfig anymore. Neither in the Network menu nor in Mail.

Do I need to create a customfeed? If so, how?

Anyway, you say: "If you do not want to wait". How long does it take for 
the modified package to show up in trunk?

Kind regards,

Suso

fededim, 2016-09-22:
Before launching "make menuconfig" execute
./scripts/feeds update -i
./scripts/install -a

this will update the feed index file ./feeds/packages.index (it's a text file and if you open it you must find emailrelay, otherwise it won't show in make menuconfig).

rodajc, 2016-09-22:
Hi Federico,

Thanks again for your reply.

After finding out what was missing in:
./scripts/*feeds *install -a

the compilation was successful. Now if I remember how I will load the 
firmware into a router and test it. I'll let you know as soon as I have 
a crash log.

Kind regards,

Suso

fededim, 2016-09-22:
Ehm are you drunk ? What did you find missing in this file ?!?

> After finding out what was missing in:
> ./scripts/*feeds *install -a
>

rodajc, 2016-09-22:
Not drunk... yet.

The *feeds* part was missing in the command line and just 
*./scripts/install -a* didn't work...

fededim, 2016-09-22:
Ah ok, sorry my fault, a typo, I edited my previous post after 20 mins

rodajc, 2016-09-27:
Hi Federico,

Please find the log file attached.

There were two messages in the spool-dir of the emailrelay POP server. 
The log file first shows the command and the output after executing.

As soon as the server is running I hit F5 in the Thunderbird mail client 
to get new messages. I insert a comment in the logfile to mark it.

The Thunderbird mail client is unable to retrieve any of the new 
messages. Instead the process stalls. Therefore I close the Thunderbird 
mail client and mark in the log file when I do this.

The remaining part of the log is the output after closing the mail 
client. The process doesn't exit properly and the listening port isn't 
closed either.

I hope you find this useful but if you want me to change the parameters 
or to test something else, please let me know.

Regards,

Suso

fededim, 2016-11-03:
Sorry for the late reply I missed your message in the email thread....I can't find any log file attached either here or with the email....where is it ?

rodajc, 2016-11-03:
Hi Federico,

I attach it again. However it is more than likely that you still can't 
see it.

You suggest that there are other methods of leaving the log. Which are 
they? I am new here, you see.

I hope you can solve the problem with emailrelay, because I'm trying to 
use Dovecot and I think it's a bit too complex for me.

Regards,

Suso

fededim, 2016-11-03:
No, you didn't attach anything, I hope you are not drunk. You have to drag the file on the browser or click "selecting them" link below the text area on GitHub. If you are unable to attach, copy and paste emailrelay log data with debug information. If instead you have switched to another package please close this issue.

rodajc, 2016-11-03:
Here you are. I attached the log in my mail client and received no complaints.
But in the browser files with extension .log are rejected. No problem: renamed it to .txt:
[emailrelay.txt](https://github.com/openwrt/packages/files/570141/emailrelay.txt)

fededim, 2016-11-03:
It seems that emailrelay has received a SIGPIPE while writing to a socket. This usually happens when you write to a socket which has been closed on the other (client) side and this usually happens when the client program doesn't wait till all the data from the server is received and simply closes the socket.  In simple words Thunderbird requests for new messages, but closes the connection before it has fully received all of them, so it seems a Thunderbird problem, which however seems very strange. To be sure enable POP3 log on Thunderbird (https://wiki.mozilla.org/MailNews:Logging) and post Thunderbird log here. Tell me also the version of Thunderbird you are using. From this log however there is no libc crash as reported in the original issue which is good. The listening port does not apply to Thunderbird, but only to emailrelay server, to which of them were you referring in the bold sentence ?

> "The Thunderbird mail client is unable to retrieve any of the new messages. Instead the process stalls. Therefore I close the Thunderbird mail client and mark in the log file when I do this. The [...]

rodajc, 2016-11-03:
The emailrelay server leaves its listening port open when I close Thunderbird.
I'm using Thunderbird 45.4.0.
Why would Thunderbird work perfectly well with other mail servers like gmail, yahoo and hotmail and suffer from a bug when retrieving
mail from the emailrelay server?

rodajc, 2016-11-07:
root@nucom:~# cat /etc/emailrelaypop.auth
`LOGIN server user1 secret`

root@nucom:~# emailrelay --pop --no-smtp --remote-clients --spool-dir /tmp/spool --pop-auth /etc/emailrelaypop.auth

me@ubuntu:~$ export NSPR_LOG_MODULES=pop3:5,timestamp
me@ubuntu:~$ thunderbird &

[tbpop3-1.txt](https://github.com/openwrt/packages/files/575915/tbpop3-1.txt)
[tbpop3-2.txt](https://github.com/openwrt/packages/files/575919/tbpop3-2.txt)
[tbpop3-3.txt](https://github.com/openwrt/packages/files/575920/tbpop3-3.txt)
[tbpop3-4.txt](https://github.com/openwrt/packages/files/575922/tbpop3-4.txt)
BTW: emailrelay server didn't crash when I closed Thunderbird client (tbpop3-4.txt).
Enjoy!

fededim, 2016-11-16:
Ok it seems that Thunderbird goes on forever downloading a message from the server after having received some of its headers (last one is at row 163 of tbpop3-1.txt), the problem is that I do not see any of the matching send logs in email relay and more emailrelay goes on error while sending data as if Thunderbird closes the connection. I have found this very strange issue of  [Thunderbird](http://kb.mozillazine.org/Performance_(Thunderbird)):

> Thunderbird appears to parse the contents of the entire Inbox folder when it downloads new mail for POP accounts. If the folder contains messages with certain missing headers (typically due to spam or a corrupted folder) it can get caught in a loop. The workaround (after first backing up your profile folder!) would be to move all of the Inbox folders messages to another folder (where you can identify and delete the troublesome messages later on) and compact the Inbox. In the worst case you might need to copy the messages in your Inbox to another folder, exit Thunderbird, back up your profile, delete both the "Inbox." and "Inbox.msf" files for that account in your profile and then run Thunderbird again. It will create a new empty Inbox folder if you do that.

Unluckily it does not state what headers drive Thunderbid crazy (I have found however this [bug](https://bugzilla.mozilla.org/show_bug.cgi?id=617923)). Please do these two other things:
1) Try to use the included new version of emailrelay in which I added further logs when sending data and attach the new emailrelay log.
2) Attach the full message received with headers by emailrelay (it is saved in the spool directory), I would like to know what is in it after the last line received by Thunderbird ("Tue Sep 27 17:12:09 CEST 2016").

**Thunderbird**

> 2016-11-07 15:55:50.854519 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: Date: Tue, 27 Sep 2016 17:12:25 +0200
> 2016-11-07 15:55:50.854528 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: User-Agent: Mozilla/5.0 (X11; Linux i686; rv:45.0) Gecko/20100101
> 2016-11-07 15:55:50.854537 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: Thunderbird/45.3.0
> 2016-11-07 15:55:50.854546 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: MIME-Version: 1.0
> 2016-11-07 15:55:50.854556 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: Content-Type: text/plain; charset=utf-8; format=flowed
> 2016-11-07 15:55:50.854565 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: Content-Transfer-Encoding: 7bit
> 2016-11-07 15:55:50.854574 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: Tue Sep 27 17:12:09 CEST 2016
> 2016-11-07 15:55:50.854583 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: 
> 2016-11-07 15:55:50.855661 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: 
> 2016-11-07 15:55:50.855677 UTC - -1216665856[b756b180]: POP3: [this=9dabb0c0] RECV: 
> 

**Emailrelay server** 

> emailrelay: info: GPop::ServerProtocol: tx>>: "+OK 2 message(s)\r\n1 emailrelay.1157.865799.59.content\r\n2 emailrelay.1157.865799.60.content\r\n."
> emailrelay: debug: GNet::Select::runOnce: detected event(s) on 1 fd(s)
> emailrelay: debug: SocketProtocolImp::readEvent: state=0
> emailrelay: info: GPop::ServerProtocol: rx<<: "RETR 1"
> emailrelay: debug: GPop::StoreLock::get: 1: /tmp/spool/emailrelay.1157.865799.59.content
> emailrelay: info: GPop::ServerProtocol: tx>>: "+OK 2643 octets"
> \>\>\>\>\> here process stalls. I close Thunderbird mail client
> emailrelay: debug: GNet::Socket::reason: error 32: Broken pipe
> emailrelay: debug: GNet::Socket::write: write error 32
> emailrelay: debug: ServerPeer::onException: exception: peer disconnected
> emailrelay: info: GPop::ServerPeer: pop connection closed: peer disconnected: 192.168.0.12:53493

[emailrelay_1.9-3_brcm63xx.ipk.zip](https://github.com/openwrt/packages/files/593470/emailrelay_1.9-3_brcm63xx.ipk.zip)

rodajc, 2016-11-19:
[emailrelay.1157.865799.zip](https://github.com/openwrt/packages/files/601741/emailrelay.1157.865799.zip)

rodajc, 2016-11-19:
root@nucom:/# opkg install /tmp/emailrelay_1.9-3_brcm63xx.ipk 
Installing emailrelay (1.9-3) to root...
Collected errors:
 * opkg_install_pkg: Package emailrelay sha256sum mismatch. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package emailrelay.

Anyway, for me it is easier if you just attach the executable or the updated sources with instructions where to put them for compilation.

fededim, 2016-11-20:
try with --force-checksum or --force-reinstall options (between "install" command and filename, see https://wiki.openwrt.org/doc/techref/opkg). You receive this error because openwrt stores the checksum of all the built packages inside an index file downloaded from package repository. Since I have changed the executable inside the package (adding new log) it is obvious that it does not match, but it is just a recoverable warning. I am unable to attach now the executable.

fededim, 2016-11-22:
Oh by the way if you are paranoid .ipk files are simply tar.gz files you can uncompress them by using tar zxvf <file .ipk>, inside you will find 2 inner tgz files (control and data), you have to uncompress again data.tar.gz and you will find emailrelay's executable under /usr/bin

rodajc, 2016-11-22:
[emailrelay2.txt](https://github.com/openwrt/packages/files/607833/emailrelay2.txt)

Something must've gone wrong as there's no debug info. I extracted the binary from the .ipk just the way you indicated. I executed it in /tmp first, then I moved it to /usr/bin and executed it again but still no debug info.
Any idea?

fededim, 2016-11-23:
In the log you posted I can't see you the additional send logs (you'd better reinstall with (opkg install <emailrelay....ipk> --force-checksum, configuration files in /etc are preserved). Anyway try this new version and let me know if it works.

[emailrelay_1.9-3_brcm63xx-patched.ipk.zip](https://github.com/openwrt/packages/files/608745/emailrelay_1.9-3_brcm63xx-patched.ipk.zip)

rodajc, 2016-11-23:
Good news. It works! This is the debug log:
[emailrelay3.txt](https://github.com/openwrt/packages/files/609893/emailrelay3.txt)
So the client now retrieves the messages from the emailrelay pop server. However, a new problem pops ;-) up. The message body is not separated from the header by a newline. Therefore it is not displayed by the client. This must be a piece of cake for you to solve.

fededim, 2016-12-02:
Now I do see the additional SocketProtocolImp logs, even though if I remember correctly I should have removed them in this last version. However there seems to be a problem while checking eof (end of file) in emailrelay source code and I have asked for some clarification directly to emailrelay developer ( https://sourceforge.net/p/emailrelay/bugs/33/ ) , I do not know if this strange behaviour is due to the different implementation c library (glibc on Linux vs MUSL on openwrt). Before venturing in the new bug let's wait for an answer from the developer. Meanwhile can you download ( https://sourceforge.net/projects/emailrelay/ ), compile (do not use binaries) and install emailrelay on your linux system and tell me if it works ? Just to be sure that these issues are happening only on Openwrt.

rodajc, 2016-12-04:
OK. But... how do I compile? I think I checked out revision 226 of the source code with this command:
svn checkout svn://svn.code.sf.net/p/emailrelay/code/trunk emailrelay-code
Where do I put this in openwrt in order to compile and where will the result end up?

fededim, 2016-12-04:
Oh no compiling in linux x86 does not need openwrt folder and it's very easy:

1) Download source code (it's the same used on openwrt version) https://sourceforge.net/projects/emailrelay/files/emailrelay/1.9/emailrelay-1.9-src.tar.gz/download
2) Uncompress source code tar zxvf <archive .tar.gz>. It will create a folder named emailrelay-1.9
3) Cd created folder emailrelay-1.9
4) Run from terminal: ./configure. Wait it finishes
5) Run from terminal: ./make. Wait it finishes
6) Done! You will find the executable under src/main/

I successfully compiled it on my Linux to test if it works ([emailrelay.zip](https://github.com/openwrt/packages/files/629792/emailrelay.zip) for linux 64bit), however all clients gave me authentication errors when fetching the email messages from emailrelay as if some authentication modules were not compiled (tried with both outlook and thunderbird and all possible authentication options), so I was unable to test the loop issue. Probably I need to reinstall linux (too much strange things happen on my configuration), in any case I asked to emailrelay developer about the issue because, even though I am not a linux C developer, it does not seem right to check for the end of file in the way it does in the source code (maybe there could be another place where emailrelay checks for eof, but from my analysis the bug is there and also on my router it seems to be fixed in that way). Let's hope for a quick reply from him.

fededim, 2016-12-05:
Just an update the way of checking eof in emailrelay is "correct", but I don't still understand why on openrwrt it loops forever. I need to test it on another router, just give me some time.

fededim, 2016-12-05:
Ok on linux works perfectly, the authentication problems were due to a bad emailrelay.auth file (do not use APOP server <user> <pwd>, but use LOGIN server <user> <pwd>). The same loop happens on another linux router brcm47xx.

fededim, 2016-12-05:
Try with this version and let me know
[emailrelay_1.9-4_brcm63xx.zip](https://github.com/openwrt/packages/files/631285/emailrelay_1.9-4_brcm63xx.zip)

rodajc, 2016-12-06:
I tried to install libstdcpp but I got a memory error. Therefore I downloaded the package first and then tried to install it. I got the same error:

root@nucom:/tmp# opkg install libstdcpp_5.3.0-1_brcm63xx.ipk 
Collected errors:
 * gz_open: fork: Out of memory.
 * pkg_init_from_file: Failed to extract control file from libstdcpp_5.3.0-1_brcm63xx.ipk.

fededim, 2016-12-07:
Ok so also the second issue resolved with the libstdc++. There is surely some problem with the openwrt standard c++ library (uclibcxx), but for now I'll switch emailrelay to use libstdc++; it takes just a little more RAM to run, but for now I did not think it is not an issue, especially in comparison to other POP3 tools.

p.s. Never seen out of memory error, maybe you have no more disk space on the first router ? Otherwise you filled your /tmp folder (it's a ram disk) and you hadn't enough memory left to uncompress the package. In latter case just try to use an usb device as storage (I do not think that you are going to run a pop3 server on a router without usb storage).

p.p.s.: test emailrelay for some time and then remember to close the issue

rodajc, 2016-12-10:
I'm testing and so far no problem. Thanks for solving this. You deserve a drink :-)
I hope you don't mind a few noob questions.
1. Does the POP server work with TLS? In the help it says:
  -K, --server-tls=<pem-file>     enables negotiated tls/ssl for smtp server
So I understand server-tls works for SMTP server but not for POP server. Is that correct?
2. Can emailrelay POP server work with encryption?
3. I know this is off topic but in my openwrt directory executing "git pull" keeps giving "Already up-to-date". How do I update my sources?
Thanks again.

fededim, 2017-01-02:
1. and 2. I do not think so, but you can directly ask to Emailrelay developer on sourceforge. For pop/imap maybe a fully featured software like Cyrus might be more suitable, unluckily no porting has been done.
3. If it shows "already up-to-date" message you have already the latest sources (sources might not change every day).

Remember to close this ticket, since the issue is fixed.

closed 2017-01-02 by hnyman

rodajc, 2018-05-05:
In server mode the --filter switch still doesn't seem to work.
Sorry it took me a few years to find out.