openwrt/packages #2406: Error creating ipv6 firewall rules

positroni, 2016-02-22:
Hi,

Today I recompiled and upgraded openwrt on my router and adblock stopped to work. It throws an error when trying to insert ipv6 rules in iptables. What I get is 

`error: failed to initialize volatile IPv6 firewall rule 'adb-forward', rc: 2`

I tried to modify the function in adblock to get the error given by ip6tables and basically it complains with the following 

`unknown reject type "icmp-port-unreachable"`

replacing with icmp6-port-unreachable seems to work but then it complains of

`unknown reject type "icmp-proto-unreachable"`

however replacing it with icmp6-proto-unreachable doesn't solve.
If I remove lines 402->410 in adblock-helper.sh then it works without problems...

dibdot, 2016-02-22:
@positroni: thanks for your report, will be fixed in 0.80.1

hnyman, 2016-02-22:
I spotted the same thing last night and mailed @dibdot about it.

The reason is the two invalid icmp6 types that you mentioned. They were introduced a few days ago when rules were tweaked for better operation with https links.

In my own router, I replaced the last rules with icmp6-addr-unreachable and that helped:
```
        f_firewall "IPv6" "filter" "A" "${adb_fwdchain_ipv6}" "adb-forward" "! -i ${adb_wandev6} -p udp -d ${adb_nullipv6} -j REJECT --reject-with icmp6-port-unreachable"
        f_firewall "IPv6" "filter" "A" "${adb_fwdchain_ipv6}" "adb-forward" "! -i ${adb_wandev6} -p tcp -d ${adb_nullipv6} -j REJECT --reject-with tcp-reset"
        f_firewall "IPv6" "filter" "A" "${adb_fwdchain_ipv6}" "adb-forward" "! -i ${adb_wandev6} -d ${adb_nullipv6} -j REJECT --reject-with icmp6-addr-unreachable"
        f_firewall "IPv6" "filter" "A" "${adb_outchain_ipv6}" "adb-output" "! -i ${adb_wandev6} -p udp -d ${adb_nullipv6} -j REJECT --reject-with icmp6-port-unreachable"
        f_firewall "IPv6" "filter" "A" "${adb_outchain_ipv6}" "adb-output" "! -i ${adb_wandev6} -p tcp -d ${adb_nullipv6} -j REJECT --reject-with tcp-reset"
        f_firewall "IPv6" "filter" "A" "${adb_outchain_ipv6}" "adb-output" "! -i ${adb_wandev6} -d ${adb_nullipv6} -j REJECT --reject-with icmp6-addr-unreachable"

```

closed 2016-02-22 by hnyman